klutchell/balena-pihole #6: Documentation clarity

Polyterative, 2019-04-04:
I was following [this guide](https://www.balena.io/blog/deploy-network-wide-ad-blocking-with-pi-hole-and-a-raspberry-pi/) and at the point of pushing this repo to the remote cloud application I got stuck. 
I was following well untill here
```
Pushing the project code to your Raspberry Pi

As you have the CLI setup and the latest code downloaded, you can now execute a single command to push that code to balenaCloud which in turn builds the Docker image and handles the process of setting it up and running it on your device.
```
The line 
```
From within the unzipped project directory, execute balena push <appName>, where appName is the application name you set back at the beginning of the guide. For example: balena push balenaPi-hole
```
was not clear at all. 
In the end it was necessary to look at the examples inside[ another page ](https://www.balena.io/docs/reference/cli/#push-applicationordevice)
```
$ balena push myApp
$ balena push myApp --source <source directory>
$ balena push myApp -s <source directory>

$ balena push 10.0.0.1
$ balena push 10.0.0.1 --source <source directory>
$ balena push 10.0.0.1 -s <source directory>
```
and perform the `$ balena push myApp --source <source directory>` command

## My final results

`C:\Users\Vlady\Desktop\balena-cli> .\balena.exe push Docker --source .\balena-pihole\`

here

[image]

[image]

I hope this helps someone else

klutchell, 2019-04-05:
Thank you for the feedback, I'll pass it on to @chrisys as the author of the guide!

chrisys, 2019-04-15:
Hey @Polyterative thanks for the feedback on the guide!

If you execute the command within the unzipped project directory the `--source` parameter is not required as the current working directory is assumed. For example if you were to go one level deeper into the `C:\Users\Vlady\Desktop\balena-cli\balena-pihole` then the command as per the guide would work (assuming you have added the balena executable to your path, as well).

Polyterative, 2019-04-16:
That makes sense. Thanks

closed 2019-04-16 by klutchell