klutchell/balena-pihole #7: Password variable

Polyterative, 2019-04-04:
I've tried to use the `WEBPASSWORD` variable but it does not appear to work

I added the variable AFTER installing the services

[image]

[image]

klutchell, 2019-04-05:
After setting the WEBPASSWORD service variable, your pihole service should have restarted and picked up the new password on startup. Did you notice if the service restarted after adding the variable?

If the service does not restart when a variable is set, you may need to open an issue on the [BalenaCloud forums](https://forums.balena.io/c/balena-cloud).

Polyterative, 2019-04-05:
It restarted but it didn't work, I also tried to restart my device and/or services with no effect

klutchell, 2019-04-05:
Okay, that's what I was going to suggest next.

I believe a recent change to Pihole means that they are now caching the webpassword after first setup, so it may be ignoring the WEBPASSWORD parameter after first boot. This will be tricky to manage via BalenaCloud since you can't even set service variables until the service has been created.

Thanks for pointing this out, I'll keep an eye on it.

chrisys, 2019-04-16:
Yeah this is a tough one to workaround. I had the same issue at one point but then it went away after subsequent deploys which fits with the caching after first setup theory.

chrisys, 2019-04-22:
I think the solution to this is just to put everything in as an environment variable, that way it can be added before the first deploy. I will update my guide and the readme to reflect this.

commit a1f9bae5 (2019-04-22): Merge pull request #8 from chrisys/patch-1

Update readme to combine service and application vars, fixes #7
commit baeb0e7e (2019-04-22): Update readme to combine service and application vars, fixes #7
closed 2019-04-22 by klutchell