klutchell/balena-pihole #9: dnscrypt-proxy does not start on balenaOS 2.32.0

chrisys, 2019-05-06:
More of an FYI for anyone having issues than anything else, but:

Due to [this bug](https://github.com/balena-io/balena-supervisor/issues/970) in the supervisor, the `dnscrypt-proxy` service won't start, which in turn means `pihole` won't function, if you're using balenaOS 2.32.0.

To work around this until it's resolved you can use v2.31.5 when adding the device if you still want to use dnscrypt, otherwise don't set `DNS1` and `DNS2` to let pihole do internal resolution and you should be fine!

This is due to the fact the image for dnscrypt uses `EXPOSE` for port 53 in its Dockerfile which is then overridden within the docker-compose.yml within this project. The bug in the supervisor prevents this functionality from working right now.

klutchell, 2019-05-07:
I also maintain the dnscrypt-proxy docker image used in this stack. Would it help if I released a new build without the EXPOSE option in the dockerfile?

chrisys, 2019-05-08:
@klutchell yup! If the `EXPOSE` is removed from the Dockerfile it should work as intended

klutchell, 2019-05-08:
Okay, the existing 2.0.23 (latest) tag has been updated to have the EXPOSE option removed. New pulls will get the fix.

closed 2019-07-15 by chrisys