klutchell/balena-pihole #10: balena push <appName> never completes

craigusus, 2019-06-28:
After following the instructions and getting to the `balena push <appName>` part. It all runs fine and I get:

`[Info]            Uploading images`
`[Success]         Successfully uploaded images`
`[Success]         Release successfully created!`
`[Info]            Release: f632086f61288803fc1ea9efaaf59317 (id: 974470)`
`[Info]            ┌────────────────┬────────────┬────────────┐`
`[Info]            │ Service        │ Image Size │ Build Time │`
`[Info]            ├────────────────┼────────────┼────────────┤`
`[Info]            │ pihole         │ 275.84 MB  │ 11 seconds │`
`[Info]            ├────────────────┼────────────┼────────────┤`
`[Info]            │ dnscrypt-proxy │ 19.06 MB   │ < 1 second │`
`[Info]            └────────────────┴────────────┴────────────┘`
`[Info]            Build finished in 22 seconds`

But it never returns to the command prompt. I can ctrl + c to cancel then I get 
`Received SIGINT, cleaning up. Please wait.`

Then if I run `balena push <appName>` again it creates a new release version, even though nothing has changed.

I am using Windows 10 and Poweshell to run the commands.

klutchell, 2019-06-28:
Hey @craigusus , I've seen this behavior before too and it's related to the balena-cli rather than this specific app stack. I thought maybe I was using it wrong :)

I'm pretty sure @chrisys will see this and help you out, but you may want to submit an issue or post on the forums at https://forums.balena.io for a faster response.

chrisys, 2019-06-28:
Hey @craigusus yup as this is a balena-cli issue please post on the forums and the cli team will see it and be able to work out what's going on. If you share the version of the cli you're using too that'll help.

closed 2019-07-13 by craigusus