klutchell/balena-pihole #15: Container continuously restarting

gomker, 2019-07-26:
The pihole container is continuously restarting. Adding a snippet, can provide full log if you think its needed. This is my first time working with BalenaCloud, let me know if any other info is needed.

Thanks

HOST OS VERSION: balenaOS 2.38.0+rev1

```
25.07.19 22:21:53 (-0500) <pihole>   [✓] DNS service is running
25.07.19 22:21:53 (-0500) <pihole>   [✓] Pi-hole blocking is Enabled
25.07.19 22:21:53 (-0500) <pihole> [cont-init.d] 20-start.sh: exited 0.
25.07.19 22:21:53 (-0500) <pihole> [cont-init.d] 30-checktft.sh: executing... 
25.07.19 22:21:53 (-0500) <pihole> ifelse: fatal: unable to exec bash
: No such file or directory
25.07.19 22:21:53 (-0500) <pihole> [cont-init.d] 30-checktft.sh: exited 111.
25.07.19 22:21:53 (-0500) <pihole> [cont-finish.d] executing container finish scripts...
25.07.19 22:21:53 (-0500) <pihole> [cont-finish.d] done.
25.07.19 22:21:53 (-0500) <pihole> [s6-finish] syncing disks.
25.07.19 22:21:59 (-0500) <pihole> [s6-finish] sending all processes the TERM signal.
25.07.19 22:22:02 (-0500) <pihole> [s6-finish] sending all processes the KILL signal and exiting.
25.07.19 22:22:04 (-0500) Service exited 'pihole sha256:e1072f6ffd746e5be8ca8d812998abee138e22fdbe24607d2cf360e61b9b9de6'
25.07.19 22:22:05 (-0500) Restarting service 'pihole sha256:e1072f6ffd746e5be8ca8d812998abee138e22fdbe24607d2cf360e61b9b9de6'
25.07.19 22:22:05 (-0500) <pihole> [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
```

klutchell, 2019-07-26:
Hey @gomker, it looks like one of our custom init scripts is failing with an obscure bash error on startup.

Did you perform a git clone on Windows to download the project? If so, it's possible that git has changed the line endings of the scripts from Unix to DOS before they were pushed to balena.

See this forum post answer where someone else had a similar issue and were able to resolve it by fixing the line endings or by downloading the source as a zip.

https://forums.balena.io/t/unable-to-get-pi-hole-working-when-following-etcher-featured-project/7327/43

gomker, 2019-07-26:
@klutchell  I did in fact do a clone in windows. I was using cygwin and on another computer with Windows bash for refrence however, my local files are still in `/mnt/c`.  I will redo with downloaded zip and let you know

gomker, 2019-07-26:
@klutchell  Thanks for the info - a clean zip download worked. 
Appreciate the quick response

closed 2019-07-26 by gomker

SturdyErde, 2020-01-14:
Hi, @gomker. I had the same problem. The clean zip download didn't do the trick for me (or I did the trick wrong). However, I followed a tip from the [thread](https://forums.balena.io/t/unable-to-get-pi-hole-working-when-following-etcher-featured-project/7327/43) that you referenced above, and simply deleted the script 30-checktft.sh from the system. It started right up after that. What are the effects of losing this file?

caiorg, 2021-01-19:
@klutchell I'm facing this issue and tried all the tips I could find with no success.
Also I'm working on a MacBook Pro with macOS Catalina, so the Windows line ending issue or other Windows-only issues, shouldn't affect.

Any clues or something I could try?

klutchell, 2021-01-19:
@caiorg could you open a new issue and include your device logs?

caiorg, 2021-01-19:
> @caiorg could you open a new issue and include your device logs?
Sure!