klutchell/balena-pihole #17: Does not work for pizero

dt-rush, 2019-09-03:
The main symptom is the container repeatedly crashing, in the logs you see:

```
cont-init.d] 20-start.sh: executing...
[cont-init.d] 20-start.sh: exited 267.
```

(among other things).

This appears to be caused by the fact that the pihole image for arm only runs on armv7. See (`pihole/docker-pihole/issues/218`](https://github.com/pi-hole/docker-pi-hole/issues/218). 

If you dive into the various related links on that issue, you can find that pihole is in a world of difficulty in trying to get an armv6-compatible image to build.

To cap it all off, even if they did produce such an image, there's still a chance things could go wrong when pulling to the rpizero, since the rpizero also - confusingly - reports:

```
# cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 000000008c985d90
```

Note CPU architecture 7. So, the rpizero may end up pulling the wrong image anyways.

sorliem, 2019-12-27:
:+1: encountered same issue

klutchell, 2019-12-27:
@dt-rush @sorliem can you try with the latest version?

I've made some changes to build dnscrypt-proxy from source so you might have better luck on other arm platforms like Pi-Zero. No guarantees since we use the [official armhf Pi-Hole](https://github.com/pi-hole/docker-pi-hole/blob/master/Dockerfile_armhf) image and have no control over the supported architectures.

BadBambi, 2020-01-03:
HI @klutchell  - thanks again for the update above, but I have encounter the same error and reviewed the latest from [https://github.com/pi-hole/docker-pi-hole](https://github.com/pi-hole/docker-pi-hole) where they have utilised the latest docker via: **image: pihole/pihole:latest** and not particularly the 4.3.2-1_armhf. 

I've tried to review both however my arch skills are somewhat lacking for any real contribution.
I have however been able to deploy the normal docker install from https://github.com/pi-hole/docker-pi-hole on the rpizero without any issues, so my next step would be to get the Balena version working as such.

Testing with the current build:
`03.01.20 10:25:20 (+0200)  pihole  [s6-init] ensuring user provided files have correct perms...exited 0.
03.01.20 10:25:20 (+0200)  pihole  [fix-attrs.d] applying ownership & permissions fixes...
03.01.20 10:25:20 (+0200)  pihole  [fix-attrs.d] 01-resolver-resolv: applying... 
03.01.20 10:25:20 (+0200)  pihole  [fix-attrs.d] 01-resolver-resolv: exited 0.
03.01.20 10:25:20 (+0200)  pihole  [fix-attrs.d] done.
03.01.20 10:25:20 (+0200)  pihole  [cont-init.d] executing container initialization scripts...
03.01.20 10:25:21 (+0200)  pihole  [cont-init.d] 20-start.sh: executing... 
03.01.20 10:25:21 (+0200)  pihole  [cont-init.d] 20-start.sh: exited 267.
03.01.20 10:25:21 (+0200)  pihole  [cont-finish.d] executing container finish scripts...
03.01.20 10:25:21 (+0200)  pihole  [cont-finish.d] done.
03.01.20 10:25:21 (+0200)  pihole  [s6-finish] syncing disks.
03.01.20 10:25:21 (+0200)  pihole  [s6-finish] sending all processes the TERM signal.
03.01.20 10:25:24 (+0200)  pihole  [s6-finish] sending all processes the KILL signal and exiting.`

Testing with the **image: pihole/pihole:latest**
`pihole  standard_init_linux.go:195: exec user process caused "exec format error"`

Anything I could perhaps try or a direction of this please?

klutchell, 2020-01-03:
Unfortunately the official Pi-Hole docker image does not support RPi Zero. I was hopeful that the changes I made could have helped with the dnscrypt-proxy container, but it seems we are still unable to support RPi Zero with the official Pi-Hole image. There's an open PR on the Pi-Hole repo to add support for `armel`

https://github.com/pi-hole/docker-pi-hole/issues/507
https://github.com/pi-hole/docker-pi-hole/pull/506

Once an official image becomes available for RPi Zero we can update this balena stack.

sorliem, 2020-01-03:
@klutchell thanks for the update. I found the same thing (they don't support armel), but I did try out https://github.com/pi-hole/docker-pi-hole/pull/506 on my rpi0 and it worked.

chrisdeely, 2020-01-29:
@sorliem are you able to provide any direction on how you used the build from  pi-hole/docker-pi-hole#506 ? I have been attempting to do the same but I am at a loss

sorliem, 2020-01-29:
@chrisdeely I replaced [this line](https://github.com/klutchell/balena-pihole/blob/3a262e0bab8299fca082b849640cca3a96be998f/pihole/Dockerfile#L14) with
```
FROM lewixlabs/pihole-pizero
```

klutchell, 2020-01-29:
I don't have a PiZero to confirm the workaround from @sorliem but I expect you will also have to change [this line](https://github.com/klutchell/balena-pihole/blob/3a262e0bab8299fca082b849640cca3a96be998f/dnscrypt-proxy/Dockerfile#L3).

~~`FROM balenalib/raspberrypi3-alpine-golang:1.12-3.10 as build`~~
`FROM balenalib/raspberry-pi-alpine-golang:1.12-3.10 as build`

chrisdeely, 2020-01-30:
@sorliem @klutchell Thank you both for the tips - applying both of those changes got me a successful build

brainstorm, 2020-04-04:
Here's a few changes to make this work on a Raspberry Pi1... I would pullrequest this, but I moved on to balena-adguard anyway on https://github.com/klutchell/balena-adguard/issues/1#issuecomment-608997037, so leaving this here in case it's useful to somebody.

Ideally this would have to be separated/refactored to work with "Multiple docker files", see https://www.balena.io/blog/running-a-desktop-manager-with-balena/ for how to pull it off @klutchell.

```diff
diff --git a/README.md b/README.md
index e7b4a54..7188bd0 100644
--- a/README.md
+++ b/README.md
@@ -17,7 +17,7 @@ To get started you'll first need to sign up for a free balenaCloud account and f

 ## Deployment

-Once your account is set up, deployment is carried out by downloading the project and pushing it to your device either via Git or the balena CLI.
+Once your account is set up, deployment is carried out by downloading the project and pushing it to your device either via Git or the balena CLI. By default raspberrypi-3 is assumed, please checkout branch `rpi1` beforehand if you with to run this on a Raspberry Pi 1.

 ### Device Variables

diff --git a/dnscrypt-proxy/Dockerfile b/dnscrypt-proxy/Dockerfile
index 6f11c1c..8092237 100644
--- a/dnscrypt-proxy/Dockerfile
+++ b/dnscrypt-proxy/Dockerfile
@@ -1,16 +1,16 @@
-FROM balenalib/raspberrypi3-alpine-golang:1.13 as build
+FROM balenalib/raspberry-pi-alpine-golang as build

-WORKDIR /go/src/github.com/DNSCrypt/dnscrypt-proxy/

-ARG DNSCRYPT_PROXY_VERSION=2.0.39
+ARG DNSCRYPT_PROXY_VERSION=2.0.42
 ARG DNSCRYPT_PROXY_URL=https://github.com/DNSCrypt/dnscrypt-proxy/archive/
-
 ENV CGO_ENABLED 0

 RUN apk add --no-cache ca-certificates curl \
        && curl -L "${DNSCRYPT_PROXY_URL}${DNSCRYPT_PROXY_VERSION}.tar.gz" -o /tmp/dnscrypt-proxy.tar.gz \
-       && tar xzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt \
-       && go build -v -ldflags="-s -w"
+       && mkdir -p /go/src/github.com/DNSCrypt && tar xvzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt
+
+WORKDIR /go/src/github.com/DNSCrypt/dnscrypt-proxy
+RUN go build -v -ldflags="-s -w"

 WORKDIR /config

diff --git a/docker-compose.yml b/docker-compose.yml
index f719a88..5fd2896 100644
--- a/docker-compose.yml
+++ b/docker-compose.yml
@@ -16,6 +16,8 @@ services:
       - '127.0.0.1'
       - '1.1.1.1'
     network_mode: host
+    depends_on:
+      - "dnscrypt-proxy"
diff --git a/README.md b/README.md
index e7b4a54..7188bd0 100644
--- a/README.md
+++ b/README.md
@@ -17,7 +17,7 @@ To get started you'll first need to sign up for a free balenaCloud account and f

 ## Deployment

-Once your account is set up, deployment is carried out by downloading the project and pushing it to your device either via Git or the ba
lena CLI.
+Once your account is set up, deployment is carried out by downloading the project and pushing it to your device either via Git or the ba
lena CLI. By default raspberrypi-3 is assumed, please checkout branch `rpi1` beforehand if you with to run this on a Raspberry Pi 1.

 ### Device Variables

diff --git a/dnscrypt-proxy/Dockerfile b/dnscrypt-proxy/Dockerfile
index 6f11c1c..8092237 100644
--- a/dnscrypt-proxy/Dockerfile
+++ b/dnscrypt-proxy/Dockerfile
@@ -1,16 +1,16 @@
-FROM balenalib/raspberrypi3-alpine-golang:1.13 as build
+FROM balenalib/raspberry-pi-alpine-golang as build

-WORKDIR /go/src/github.com/DNSCrypt/dnscrypt-proxy/

-ARG DNSCRYPT_PROXY_VERSION=2.0.39
+ARG DNSCRYPT_PROXY_VERSION=2.0.42
 ARG DNSCRYPT_PROXY_URL=https://github.com/DNSCrypt/dnscrypt-proxy/archive/
-
 ENV CGO_ENABLED 0

 RUN apk add --no-cache ca-certificates curl \
        && curl -L "${DNSCRYPT_PROXY_URL}${DNSCRYPT_PROXY_VERSION}.tar.gz" -o /tmp/dnscrypt-proxy.tar.gz \
-       && tar xzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt \
-       && go build -v -ldflags="-s -w"
+       && mkdir -p /go/src/github.com/DNSCrypt && tar xvzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt
+
+WORKDIR /go/src/github.com/DNSCrypt/dnscrypt-proxy
+RUN go build -v -ldflags="-s -w"

 WORKDIR /config

diff --git a/docker-compose.yml b/docker-compose.yml
index f719a88..5fd2896 100644
--- a/docker-compose.yml
+++ b/docker-compose.yml
@@ -16,6 +16,8 @@ services:
       - '127.0.0.1'
       - '1.1.1.1'
     network_mode: host
+    depends_on:
+      - "dnscrypt-proxy"

   # set pihole DNS1 and DNS2 service variables to 127.0.0.1#5053 to use dnscrypt-proxy for upstream DNS
   dnscrypt-proxy:
diff --git a/pihole/Dockerfile b/pihole/Dockerfile
index c46a631..f9c2a12 100644
--- a/pihole/Dockerfile
+++ b/pihole/Dockerfile
@@ -1,8 +1,8 @@
-FROM balenalib/raspberrypi3-debian:jessie-build AS build
+FROM balenalib/raspberry-pi-debian:bullseye AS build

 WORKDIR /usr/src

-RUN install_packages cmake libraspberrypi-dev
+RUN install_packages cmake automake autoconf build-essential libraspberrypi-dev

 RUN curl -sSL https://raw.githubusercontent.com/tasanakorn/rpi-fbcp/master/CMakeLists.txt -O
 RUN curl -sSL https://raw.githubusercontent.com/tasanakorn/rpi-fbcp/master/main.c -O
@@ -11,7 +11,9 @@ WORKDIR /usr/src/build

 RUN cmake .. && make

-FROM pihole/pihole:v4.4-armhf
+RUN apt-get purge -y automake autoconf build-essential && rm -rf /var/cache/apt/archives/*
+
+FROM pihole/pihole:v4.4-armel

 WORKDIR /usr/src

@@ -28,5 +30,5 @@ RUN sed -i '/$AUTHORIZED_HOSTNAMES = array(/ a "balena-devices.com",' /var/www/h
 RUN echo "bind-interfaces" >> /etc/dnsmasq.conf

 SHELL ["/bin/bash", "-o", "pipefail", "-c"]
-RUN curl -fsSL https://github.com/jpmck/PADD/archive/v3.0.2.tar.gz | tar xz --strip-components 1 \
+RUN curl -fsSL https://github.com/jpmck/PADD/archive/v3.1.tar.gz | tar xz --strip-components 1 \
        && chmod +x padd.sh
```

klutchell, 2020-08-10:
Support for Raspberry Pi Zero is in `Dockerfile.rpi` so by renaming it to `Dockerfile` it should work as expected.

closed 2020-08-10 by klutchell