klutchell/balena-pihole #18: Currently broken on rpi3

dt-rush, 2019-09-03:
I tried to deploy on rpi3 after the issues with rpizero, and I've run into what is apparently an issue [others have seen in the forums](). 

```
::: Testing pihole-FTL DNS: sudo: unable to resolve host f61e339

dnsmasq: unknown interface eth0
[cont-init.d] 20-start.sh: exited 1.
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] syncing disks.
[s6-finish] sending all processes the TERM signal.
```

I did set `DNSMASQ_LISTENING=wlan0` as well, since I'm on wifi, to try to resolve it, as mentioned in the forum post, but no change.

dt-rush, 2019-09-04:
Setting `INTERFACE` alongside `DNSMASQ_LISTENING` both to the interface I wished to target worked. I can see in the README that you list these variables but it's not clear that we need to define them ourselves. Mind if I push a PR on the README to clarify?

dt-rush, 2019-09-04:
The blog post related to this *does* specify this need, but I wasn't following the blog post :P 

https://www.balena.io/blog/deploy-network-wide-ad-blocking-with-pi-hole-and-a-raspberry-pi/#2setconfigurationoptions

klutchell, 2019-09-04:
Go ahead an open a pull request if you have suggested changes to the README.

closed 2019-10-03 by klutchell