klutchell/balena-pihole #21: dnscrypt-proxy stopped working since last update

tsialex, 2019-10-02:
I've updated to the yesterdays release with the changed environment variables/new dnscrypt-proxy and noticed that dnscrypt-proxy stopped working. 

Environment variables updated and even with a brand new install, dnscrypt-proxy still don't work.

Could you please check it? Thanks in advance.

klutchell, 2019-10-02:
Hmm, I didn't notice the dnscrypt-service was restarting but I see now that it is very unstable. Is that the same thing you're seeing, just the service restarting with no logs?

I've rolled back the last 3 commits, so you'll have to undo some of your ENV var changes but it appears to be stable again.

I'll have a look at the recent versions of dnscrypt-proxy and see if I can't determine what's going wrong.

tsialex, 2019-10-03:
Thanks. Yes, from Balena console I see that dnscrypt-proxy is downloaded/installed but never shows running and no logs whatsoever. 

With up to September 25 commits and previous environment variables, everything is working as expected.

klutchell, 2019-10-03:
UPDATE: the latest build of https://github.com/klutchell/dnscrypt-proxy has been fixed upstream and both services appear stable

Thank you for bringing this to my attention. Let me know if it works for you after an update and we can close this issue.

tsialex, 2019-10-03:
Thx, seems to be working again. You can close this issue.

closed 2019-10-03 by klutchell