klutchell/balena-pihole #22: Web panel service not working

jw910731, 2019-10-11:
The lighttpd service didn't started porperly. I manually start it up and the web page isn't able to show up properly. After trying to access the admin page, lighttpd will stop by itself.
Here's the log(`/var/log/lighttpd/error.log`):
```
2019-10-11 18:49:08: (log.c.217) server started
2019-10-11 18:50:41: (server.c.1534) logfiles cycled UID = 0 PID = 893
2019-10-11 18:50:41: (mod_fastcgi.c.2424) unexpected end-of-file (perhaps the fastcgi process died): pid: 525 socket: unix:/var/run/lighttpd/php.socket-0
2019-10-11 18:50:41: (mod_fastcgi.c.3175) response not received, request sent: 1004 on socket: unix:/var/run/lighttpd/php.socket-0 for /admin/index.php?login, closing connection
2019-10-11 18:50:41: (mod_fastcgi.c.2424) unexpected end-of-file (perhaps the fastcgi process died): pid: 525 socket: unix:/var/run/lighttpd/php.socket-0
2019-10-11 18:50:41: (mod_fastcgi.c.3175) response not received, request sent: 993 on socket: unix:/var/run/lighttpd/php.socket-0 for /admin/index.php?, closing connection
2019-10-11 18:50:41: (server.c.1828) server stopped by UID = 0 PID = 893
```

I'd viewed [This solution](https://discourse.pi-hole.net/t/500-internal-server-error-pihole-seems-to-be-working/12972/13). But in my case the php didn't give me the php fatal error.

klutchell, 2019-10-12:
What is the output of `lighttpd -t -f /etc/lighttpd/lighttpd.conf` when run inside the pihole container? You may want to also ask on the [pihole forums](https://discourse.pi-hole.net/c/bugs-problems-issues) as it's not something I'm able to reproduce.

klutchell, 2019-11-04:
Are you still experiencing this issue?

closed 2020-08-10 by klutchell