klutchell/balena-pihole #23: Can't push release over balenaCloud as dnscrypt-proxy image doesn't exist

flaprimo, 2019-11-04:
### The problem
Following [balenaCloud pihole tutorial](https://www.balena.io/blog/deploy-network-wide-ad-blocking-with-pi-hole-and-a-raspberry-pi/), I got stuck at "3. Deploy the software > 3. Pushing the project code to your Raspberry Pi".
The error is:
```
[Error] An error occured: (HTTP code 404) no such image - no such image: klutchell/dnscrypt-proxy:arm32v7-2.0.27: No such image: klutchell/dnscrypt-proxy:arm32v7-2.0.27
```

### The solution
Change `balena-pihole/docker-compose.yml` (line 23):
`image: klutchell/dnscrypt-proxy:arm32v7-2.0.27` -> `image: klutchell/dnscrypt-proxy`

Change `balena-pihole/pihole/Dockerfile` (line 14):
`FROM pihole/pihole:4.3.2-1_armhf` -> `FROM pihole/pihole`

### Device configuration
Host OS Version: balenaOS 2.38.0+rev1 production
Supervisor Version: 9.15.7
Type: Raspberry Pi 3 (using 64bit OS) (BETA)

klutchell, 2019-11-04:
I pushed a fix that resolves the missing image error. Let me know if this works for you!

flaprimo, 2019-11-04:
Thank you, just pushed the release and it works fine!

closed 2019-11-04 by flaprimo