klutchell/balena-pihole #24: dnscrypt-proxy stopped working after yesterday update (1576f53814b75106539ceca9c4766d553cb8732c)

tsialex, 2019-11-05:
Seems no one reported this yet, I updated to the last commit today and noticed that dnscrypt-proxy is exiting and restarting constantly:

> 05.11.19 15:59:22 (-0300) Restarting service 'dnscrypt-proxy sha256:27c95604f138ab1f461d66048be5ede0b222bf11ea3f8fe81aaeeebfd43d9bfa'
> 05.11.19 15:59:22 (-0300) Service exited 'dnscrypt-proxy sha256:27c95604f138ab1f461d66048be5ede0b222bf11ea3f8fe81aaeeebfd43d9bfa'
> 05.11.19 15:58:43 (-0300)  dnscrypt-proxy  standard_init_linux.go:207: exec user process caused "exec format error"
> 05.11.19 15:59:17 (-0300)  dnscrypt-proxy  standard_init_linux.go:207: exec user process caused "exec format error"

klutchell, 2019-11-05:
Which platform are you running on (could you share the output of `uname -a`)?

tsialex, 2019-11-05:
Sorry for not provide this info before, I'm using the current balena-cloud and the same problem happens with Pi2B and Pi3B.

> =============================================================
    Welcome to balenaOS
=============================================================
root@0f35d7f:~# uname -a
Linux 0f35d7f 4.19.71 #1 SMP Tue Sep 17 09:52:03 UTC 2019 armv7l armv7l armv7l GNU/Linux
root@0f35d7f:~#

klutchell, 2019-11-05:
I think I see the problem. The balena builders are arm64 by default, so that's what was being pulled from the docker repo. I've pushed https://github.com/klutchell/balena-pihole/commit/432298adfbc77ea2d53d02d072ddee13c3a5dc16 to force the platform. at build time.

It works on my Pi4. Can you try the change?

tsialex, 2019-11-05:
> I think I see the problem. The balena builders are arm64 by default, so that's what was being pulled from the docker repo. I've pushed [432298a](https://github.com/klutchell/balena-pihole/commit/432298adfbc77ea2d53d02d072ddee13c3a5dc16) to force the platform at build time.
> 
> It works on my Pi4. Can you try the change?

Sure, I'll test and report back.

tsialex, 2019-11-05:
dnscrypt-proxy is now loading correctly and seems to work, thx. I'll close the issue.

closed 2019-11-05 by tsialex