klutchell/balena-pihole #25: PADD does not work with pitft 2.8"

drew-kun, 2019-11-30:
Following [these instructions](https://www.balena.io/blog/add-a-display-to-your-pi-hole-for-monitoring-and-stats/) but using [adafruit 2.8" capacitive pitft display](https://www.adafruit.com/product/1983) instead. 

Here are the instructions for enabling the mentioned pitft display: https://github.com/balena-io-projects/balena-wpe#capacitive-touchscreen

The display seems to work well after adding the following balena vars:

```
RESIN_HOST_CONFIG_gpu_mem  =  64
RESIN_HOST_CONFIG_dtoverlay  =  pitft28-capacitive,rotate=270,speed=25000000,fps=20
RESIN_HOST_CONFIG_hdmi_cvt  =  320 240 60 1 0 0 0
RESIN_HOST_CONFIG_hdmi_force_hotplug  = 1
RESIN_HOST_CONFIG_hdmi_group  = 4
RESIN_HOST_CONFIG_hdmi_mode  =  87
```

After booting pi the screen is white with the Balena Logo, then after I see in logs `Starting PADD` the screen turns black, but no text is visible.

Please help to make PADD work with the mentioned display.
Thank you.

klutchell, 2019-12-03:
Have you tried with any other values? I find that depending on the screen it can require some trial and error. Here are the suggestions from the readme you linked.

Key | Value
-- | --
RESIN_HOST_CONFIG_hdmi_force_hotplug | 1
RESIN_HOST_CONFIG_hdmi_group | 2
RESIN_HOST_CONFIG_hdmi_mode | 87
RESIN_HOST_CONFIG_hdmi_cvt | 320 240 60 1 0 0 0

So changing `RESIN_HOST_CONFIG_hdmi_group` from 4 to 2 might help, but I don't know what else you've tried.

drew-kun, 2019-12-04:
Thanks for your reply! I have already created the [issue](https://forums.balena.io/t/padd-from-balena-pihole-project-does-not-work-with-pitft-2-8/48049/5) on balena forum and seems like it's a balena version problem. Here is the related issue for tracking: https://github.com/balena-os/meta-balena/issues/1772

chrisys, 2019-12-04:
Just to make things clearer, PADD is also broken on the 3.5" PiTFT when using latest balenaOS too (and probably all other variants where fbcp is involved).

closed 2020-10-13 by klutchell