klutchell/balena-pihole #26: Activate DHCP on first boot

Xyala, 2019-12-05:
Thanks for sharing this, great project and it works amazing!

Quick question, I'm using the DHCP from the PiHole and I would like to see if there is a way to have it turned ON by default so that it's enabled straight from the start on first boot of the balena app.

And, of course, if there is a way to pre-set the DHCP settings (ip range, etc.)

Thanks!

klutchell, 2019-12-06:
Is that something that is possible with the Pi-hole service itself, or the upstream image?

I would check the [Docker Pi-hole](https://github.com/pi-hole/docker-pi-hole/) repo or the [Pi-hole Userspace](https://discourse.pi-hole.net/) to see if that is possible with command line arguments.

Xyala, 2019-12-06:
Good question, but if it would be possible it would be great.

I looked around the Docker Pi-Hole but if there is something about how it can be done it isn't clearly stated how. 

I'm going to post on the Userspace forums. 
Thanks!

Xyala, 2019-12-06:
Link to the forum : https://discourse.pi-hole.net/t/enable-and-set-pihole-dhcp-server-at-container-start/25894/3

I'm invited to reach out to who wrote the code haha.

DigeratiDoge, 2020-01-11:
In CLI, you can use the enabledhcp command.

sudo pihole -a enabledhcp [from] [to] [router] [leasetime] [domain]

klutchell, 2020-01-11:
Great, thanks @DigeratiDoge !

@Xyala you can try adding something like this as a final RUN command in your `pihole/Dockerfile` with your own values and see if you get the behavior you're looking for.

```bash
RUN pihole -a enabledhcp [from] [to] [router] [leasetime] [domain]
```

closed 2020-08-10 by klutchell