klutchell/balena-pihole #27: Orange Pi Plus Dockerfile parse error line 1: Unknown flag: platform

ZenBoy999, 2019-12-24:
# DEBUG=1  ../balena-cli/balena push 192.168.7.75
[debug] original argv0="../balena-cli/balena" argv=[/root/balena-cli/balena,/snapshot/versioned-source/bin/balena,push,192.168.7.75] length=4
[debug] Using . as build source
[Debug]   Checking we can access device
[Debug]   Sending request to http://192.168.7.75:48484/ping
[Debug]   Checking device version: 9.11.3
[Info]    Starting build on device 192.168.7.75
[Debug]   Loading project...
[Debug]   Resolving project...
[Info]    Compose file detected
[Debug]   Creating project...
[Debug]   Tarring all non-ignored files...
[Debug]   Sending request to http://192.168.7.75:48484/v2/local/device-info
[Debug]   Found build tasks:
[Debug]       pihole: build [./pihole]
[Debug]       dnscrypt-proxy: build [./dnscrypt-proxy]
[Debug]   Resolving services with [orange-pi-zero|armv7hf]
[Debug]   Found project types:
[Debug]       pihole: Standard Dockerfile
[Debug]       dnscrypt-proxy: Standard Dockerfile
[Debug]   Probing remote daemon for cache images
[Debug]   Using 19 on-device images for cache...
[Debug]   Starting builds...
[Build]   [pihole] Step 1/18 : FROM balenalib/raspberrypi3-debian:jessie-build AS build
[Build]   [pihole]  ---> 17662902215f
[Build]   [pihole] Step 2/18 : WORKDIR /usr/src
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> ac19bbf6da5b
[Build]   [pihole] Step 3/18 : RUN install_packages cmake libraspberrypi-dev
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> 9394463b9b3a
[Build]   [pihole] Step 4/18 : RUN curl -sSL https://raw.githubusercontent.com/tasanakorn/rpi-fbcp/master/CMakeLists.txt -O
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> d5f7753147a6
[Build]   [pihole] Step 5/18 : RUN curl -sSL https://raw.githubusercontent.com/tasanakorn/rpi-fbcp/master/main.c -O
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> 9850ced08904
[Build]   [pihole] Step 6/18 : WORKDIR /usr/src/build
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> 21f0839c213c
[Build]   [pihole] Step 7/18 : RUN cmake .. && make
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> 44cc809a7173
[Build]   [pihole] Step 8/18 : FROM pihole/pihole:4.3.2-1_armhf
[Build]   [pihole]  ---> 1ad60393d9c2
[Build]   [pihole] Step 9/18 : WORKDIR /usr/src
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> c778e3f9888a
[Build]   [pihole] Step 10/18 : COPY --from=build /opt/vc/lib/* /opt/vc/lib/
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> bca8c01ef88d
[Build]   [pihole] Step 11/18 : COPY --from=build /usr/src/build/fbcp /usr/src/
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> 9080fb6e9035
[Build]   [pihole] Step 12/18 : COPY services/ /etc/services.d/
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> 96436e73d341
[Build]   [pihole] Step 13/18 : COPY init/ /etc/cont-init.d/
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> c06f133b63bb
[Build]   [pihole] Step 14/18 : RUN sed -i '/$AUTHORIZED_HOSTNAMES = array(/ a "balena-devices.com",' /var/www/html/admin/scripts/pi-hole/php/auth.php
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> a96f75c25b84
[Build]   [pihole] Step 15/18 : RUN echo "bind-interfaces" >> /etc/dnsmasq.conf
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> 680e88eb288f
[Build]   [pihole] Step 16/18 : RUN curl -fsSL https://github.com/jpmck/PADD/archive/v3.0.2.tar.gz | tar xz --strip-components 1         && chmod +x padd.sh
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> 0ca866d0e138
[Build]   [pihole] Step 17/18 : RUN curl -sSL https://v.firebog.net/hosts/lists.php?type=tick -o /etc/pihole/adlists.list
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> b7229e0cd151
[Build]   [pihole] Step 18/18 : LABEL "io.resin.local.image"='1' "io.resin.local.service"='pihole'
[Build]   [pihole]  ---> Using cache
[Build]   [pihole]  ---> 24984605b959
[Build]   [pihole] Successfully built 24984605b959
[Build]   [pihole] Successfully tagged local_image_pihole:latest
Some services failed to build:
        dnscrypt-proxy: (HTTP code 400) unexpected - Dockerfile parse error line 1: Unknown flag: platform

If you need help, don't hesitate in contacting our support forums at
https://forums.balena.io

For CLI bug reports or feature requests, have a look at the GitHub issues or
create a new one at: https://github.com/balena-io/balena-cli/issues/

ZenBoy999, 2019-12-24:
Hardware: Orange Pi Zero 
[image]

klutchell, 2019-12-24:
It appears the docker (balena) daemon on your device doesn't support the `--platform=linux/arm/v6` flag. I think if you will be building directly on the target device you can remove `--platform=linux/arm/v6` from the dnscrypt-proxy Dockerfile and hopefully the correct image will be downloaded automatically for your architecture.

However, there is also a known issue with Pi Zero (and maybe Orange Pi Zero) that it identifies as armv7 even though it requires armv6hf architecture images. You may run into that, you may not. Let us know.

klutchell, 2019-12-27:
@ZenBoy999  can you try with the latest version?

I've made some changes to build dnscrypt-proxy from source so you might have better luck on other arm platforms like Orange Pi. No guarantees since we use the [official armhf Pi-Hole](https://github.com/pi-hole/docker-pi-hole/blob/master/Dockerfile_armhf) image and have no control over the supported architectures.

ZenBoy999, 2019-12-28:
Cool! It works! 
[image]

closed 2019-12-28 by ZenBoy999