klutchell/balena-pihole #29: Unable to build PiHole: E: Unable to locate package libraspberrypi-dev

CyberEditz, 2019-12-29:
RPi3
```
..\balena.exe push RPi3-Home --verbose
[Info]            Starting build for RPi3-Home, user *********
[Info]            Dashboard link: https://dashboard.balena-cloud.com/apps/*******/devices
[Info]            Building on arm03
[Info]            Pulling previous images for caching purposes...
[Success]         Successfully pulled cache images
[pihole]          Step 1/17 : FROM balenalib/raspberrypi3-64-debian:jessie-build AS build
[dnscrypt-proxy]  Step 1/17 : FROM balenalib/raspberrypi3-64-alpine-golang:1.12-3.10 as build
[pihole]           ---> b9ff226e22de
[pihole]          Step 2/17 : WORKDIR /usr/src
[pihole]          Using cache
[pihole]           ---> 2c2dd19121fd
[pihole]          Step 3/17 : RUN install_packages cmake libraspberrypi-dev
[dnscrypt-proxy]   ---> 3a3c7fccdf78
[dnscrypt-proxy]  Step 2/17 : WORKDIR /go/src/github.com/DNSCrypt/dnscrypt-proxy/
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 27ae110cba1f
[dnscrypt-proxy]  Step 3/17 : ARG DNSCRYPT_PROXY_VERSION=2.0.36
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 2666cbd581fd
[dnscrypt-proxy]  Step 4/17 : ARG DNSCRYPT_PROXY_URL=https://github.com/DNSCrypt/dnscrypt-proxy/archive/  
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 6447963685fd
[dnscrypt-proxy]  Step 5/17 : ENV CGO_ENABLED 0
[pihole]           ---> Running in 099286f44384
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> e65842ed87fa
[dnscrypt-proxy]  Step 6/17 : RUN apk add --no-cache ca-certificates curl       && curl -L "${DNSCRYPT_PROXY_URL}${DNSCRYPT_PROXY_VERSION}.tar.gz" -o /tmp/dnscrypt-proxy.tar.gz  && tar xzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt    && go build -v -ldflags="-s -w"
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 557da6c7c3f0
[dnscrypt-proxy]  Step 7/17 : WORKDIR /config
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 3dca8ae10ee2
[dnscrypt-proxy]  Step 8/17 : RUN cp -a /go/src/github.com/DNSCrypt/dnscrypt-proxy/example-* ./
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 2b121c9f2419
[dnscrypt-proxy]  Step 9/17 : COPY dnscrypt-proxy.toml ./
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 5a260ff5b4b9
[dnscrypt-proxy]  Step 10/17 : FROM scratch
[dnscrypt-proxy]   --->
[dnscrypt-proxy]  Step 11/17 : COPY --from=build /etc/passwd /etc/group /etc/
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 90195e931997
[dnscrypt-proxy]  Step 12/17 : COPY --from=build /etc/ssl/certs/ca-certificates.crt /etc/ssl/certs/       
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 07c405aac592
[dnscrypt-proxy]  Step 13/17 : COPY --from=build /go/src/github.com/DNSCrypt/dnscrypt-proxy/dnscrypt-proxy /usr/local/bin/
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 8e157ad3f46d
[dnscrypt-proxy]  Step 14/17 : COPY --from=build --chown=nobody:nogroup /config /config
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 3c53cfc2573a
[dnscrypt-proxy]  Step 15/17 : USER nobody
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> d40ca6ffab72
[dnscrypt-proxy]  Step 16/17 : ENTRYPOINT ["dnscrypt-proxy", "-config", "/config/dnscrypt-proxy.toml"]    
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 38dfbf27a58d
[dnscrypt-proxy]  Step 17/17 : RUN ["dnscrypt-proxy", "-version"]
[dnscrypt-proxy]  Using cache
[dnscrypt-proxy]   ---> 0fcbe8d1aee4
[dnscrypt-proxy]  Successfully built 0fcbe8d1aee4
[pihole]          Reading package lists...
[pihole]          Building dependency tree...
[pihole]          Reading state information...
[pihole]          E: Unable to locate package libraspberrypi-dev
[pihole]
[pihole]          apt failed, retrying
[pihole]          Reading package lists...
[pihole]          Building dependency tree...
[pihole]          Reading state information...
[pihole]          E: Unable to locate package libraspberrypi-dev
[pihole]
[pihole]          apt failed, retrying
[pihole]          Reading package lists...
[pihole]          Building dependency tree...
[pihole]          Reading state information...
[pihole]          E: Unable to locate package libraspberrypi-dev
[pihole]
[pihole]          Removing intermediate container 099286f44384
[pihole]          The command '/bin/sh -c install_packages cmake libraspberrypi-dev' returned a non-zero code: 100
[Info]            Uploading images
[Success]         Successfully uploaded images
[Error]           Some services failed to build:
[Error]             Service: pihole
[Error]               Error: The command '/bin/sh -c install_packages cmake libraspberrypi-dev' returned a non-zero code: 100
[Info]            Built on arm03
[Error]           Not deploying release.
Remote build failed

Additional information may be available in debug mode. Prefix the command
line with DEBUG=1, i.e.:  DEBUG=1 balena ...
If you need help, don't hesitate in contacting our support forums at
https://forums.balena.io

For bug reports or feature requests, have a look at the GitHub issues or
create a new one at: https://github.com/balena-io/balena-cli/issues/
```

klutchell, 2019-12-29:
Unfortunately this stack does not currently work on 64-bit, as there are some dependencies to build PADD that are not compatible. If you use a 32-bit app it should work!

CyberEditz, 2019-12-29:
Okay, let me re-flash my SD! I'll give an update when done!

CyberEditz, 2019-12-29:
So I re-flashed my RPi3 with the normal device type, still got this:
`[pihole]          Reading package lists...
[pihole]          Building dependency tree...
[pihole]          Reading state information...
[pihole]          E: Unable to locate package libraspberrypi-dev
[pihole]
[pihole]          apt failed, retrying
[pihole]          Reading package lists...
[pihole]          Building dependency tree...
[pihole]          Reading state information...
[pihole]          E: Unable to locate package libraspberrypi-dev
[pihole]
[pihole]          apt failed, retrying
[pihole]          Reading package lists...
[pihole]          Building dependency tree...
[pihole]          Reading state information...
[pihole]          E: Unable to locate package libraspberrypi-dev
[pihole]
[pihole]          Removing intermediate container 24a215d198ca
[pihole]          The command '/bin/sh -c install_packages cmake libraspberrypi-dev' returned a non-zero code: 100
[Info]            Uploading images
[Success]         Successfully uploaded images
[Error]           Some services failed to build:
[Error]             Service: pihole
[Error]               Error: The command '/bin/sh -c install_packages cmake libraspberrypi-dev' returned a non-zero code: 100
[Info]            Built on arm01
[Error]           Not deploying release.
Remote build failed`

CyberEditz, 2019-12-29:
Awesome, great thanks, it is working now on 32 bit!

closed 2019-12-29 by CyberEditz