klutchell/balena-pihole #34: Error building dnscrypt-proxy

westhomas, 2020-01-24:
Following https://www.balena.io/blog/deploy-network-wide-ad-blocking-with-pi-hole-and-a-raspberry-pi/

Raspberry Pi 4 B
balenaOS 2.46.1+rev3

Balena CLI (installed with brew)
```
$ balena version
11.21.7
```

```
$ git log
3a262e0 (HEAD -> master, origin/master, origin/HEAD) update readme
```

```
$ balena push Pi-hole
[Info]            Starting build for Pi-hole, user g_wesley_thomas
[Info]            Dashboard link: https://dashboard.balena-cloud.com/apps/1572884/devices
[Info]            Building on arm03
[Info]            Pulling previous images for caching purposes...
[Success]         Successfully pulled cache images
[pihole]          Step 1/17 : FROM balenalib/raspberrypi3-debian:jessie-build AS build
[dnscrypt-proxy]  Step 1/17 : FROM balenalib/raspberrypi3-alpine-golang:1.12-3.10 as build
[pihole]           ---> fb41fe462cd7
[pihole]          Step 2/17 : WORKDIR /usr/src
[dnscrypt-proxy]   ---> 88b016536cff
[dnscrypt-proxy]  Step 2/17 : WORKDIR /go/src/github.com/DNSCrypt/dnscrypt-proxy/
[pihole]           ---> Running in 726008c13c54
[dnscrypt-proxy]   ---> Running in 82ddb8dd603d
[pihole]          Removing intermediate container 726008c13c54
[pihole]           ---> b75e7615c4be
[pihole]          Step 3/17 : RUN install_packages cmake libraspberrypi-dev
[dnscrypt-proxy]  Removing intermediate container 82ddb8dd603d
[dnscrypt-proxy]   ---> 45b3fb9ffd4d
[dnscrypt-proxy]  Step 3/17 : ARG DNSCRYPT_PROXY_VERSION=2.0.36
[pihole]           ---> Running in 065e7e622735
[dnscrypt-proxy]   ---> Running in e23a25666877
[dnscrypt-proxy]  Removing intermediate container e23a25666877
[dnscrypt-proxy]   ---> 76a501beab69
[dnscrypt-proxy]  Step 4/17 : ARG DNSCRYPT_PROXY_URL=https://github.com/DNSCrypt/dnscrypt-proxy/archive/
[dnscrypt-proxy]   ---> Running in fa5a54a6335e
[dnscrypt-proxy]  Removing intermediate container fa5a54a6335e
[dnscrypt-proxy]   ---> bd11b8976ca0
[dnscrypt-proxy]  Step 5/17 : ENV CGO_ENABLED 0
[dnscrypt-proxy]   ---> Running in c830dd2973b8
[dnscrypt-proxy]  Removing intermediate container c830dd2973b8
[dnscrypt-proxy]   ---> 660c05ff8c91
[dnscrypt-proxy]  Step 6/17 : RUN apk add --no-cache ca-certificates curl 	&& curl -L "${DNSCRYPT_PROXY_URL}${DNSCRYPT_PROXY_VERSION}.tar.gz" -o /tmp/dnscrypt-proxy.tar.gz 	&& tar xzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt 	&& go build -v -ldflags="-s -w"
[dnscrypt-proxy]   ---> Running in a8656cf20e66
[dnscrypt-proxy]  fetch http://dl-cdn.alpinelinux.org/alpine/v3.10/main/armv7/APKINDEX.tar.gz
[dnscrypt-proxy]  fetch http://dl-cdn.alpinelinux.org/alpine/v3.10/community/armv7/APKINDEX.tar.gz
[dnscrypt-proxy]  OK: 95 MiB in 80 packages
[dnscrypt-proxy]    % Total    % Received % Xferd  Average Spee
[dnscrypt-proxy]  d   Time    Time     Time  Current
[dnscrypt-proxy]
[dnscrypt-proxy]
[dnscrypt-proxy]         Dload
[dnscrypt-proxy]   Upload   Total
[dnscrypt-proxy]   Spent    Le
[dnscrypt-proxy]  ft  Speed
[dnscrypt-proxy]
  0     0-proxy]
[dnscrypt-proxy]      0
[dnscrypt-proxy]      0
[dnscrypt-proxy]   0
[dnscrypt-proxy]   0
[dnscrypt-proxy]   0
[dnscrypt-proxy]     0 -
[... 138 lines cut ...]
[pihole]          Removing intermediate container 6b20276f6c5c
[pihole]           ---> 54bfb12190b6
[pihole]          Successfully built 54bfb12190b6
[Info]            Uploading images
[Success]         Successfully uploaded images
[Error]           Some services failed to build:
[Error]             Service: dnscrypt-proxy
[Error]               Error: The command '/bin/sh -c apk add --no-cache ca-certificates curl 	&& curl -L "${DNSCRYPT_PROXY_URL}${DNSCRYPT_PROXY_VERSION}.tar.gz" -o /tmp/dnscrypt-proxy.tar.gz 	&& tar xzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt 	&& go build -v -ldflags="-s -w"' returned a non-zero code: 56
[Info]            Built on arm03
[Error]           Not deploying release.
Remote build failed

Additional information may be available by setting a DEBUG=1 environment
variable: "set DEBUG=1" on a Windows command prompt, or "export DEBUG=1"
on Linux or macOS.

If you need help, don't hesitate in contacting our support forums at
https://forums.balena.io

For CLI bug reports or feature requests, have a look at the GitHub issues or
create a new one at: https://github.com/balena-io/balena-cli/issues/

(node:20892) [DEP0066] DeprecationWarning: OutgoingMessage.prototype._headers is deprecated

```

westhomas, 2020-01-24:
I did an `export DEBUG=1` and then reran the `balena push Pi-hole` and it seemed to work this time.   ¯\_(ツ)_/¯

klutchell, 2020-01-24:
Yeah, I just tried it myself at it seems to be working. I think the CURL command to download the dnscrypt-proxy source from github flaked out. I could add a retry there to prevent this in the future but I don't think it's likely to happen again.

Are we good to close?

closed 2020-01-24 by westhomas