klutchell/balena-pihole #35: Can't get into dnscrypt-proxy container

drew-kun, 2020-02-09:
When trying to get into the dnscrypt-proxy container getting:

```
$ balena ssh 192.168.2.2 dnscrypt-proxy
OCI runtime exec failed: exec failed: container_linux.go:345: starting container process caused "exec: \"/bin/sh\": stat /bin/sh: no such file or directory": unknown
Connection to 192.168.2.2 closed.
```

Similar results in BalenaCloud shell:
[image]

Is it supposed to be possible to get there?

klutchell, 2020-02-09:
That is by design. By not having a shell in the `scratch` base image we are lowering the possible attack surfaces of the container.

If you want to poke around in the dnscrypt-proxy container, try changing `FROM scratch` to `FROM alpine` in `dnscrypt-proxy/Dockerfile`.

drew-kun, 2020-02-09:
Oh, I see! Thank you very much for explanation! Closing then..

closed 2020-02-09 by drew-kun