klutchell/balena-pihole #37: Can't push to my application with default files.

Vysp3r, 2020-04-06:
I just installed balenaOS and successfully installed balenaSound and wanted to install Pi-hole to so I followed the tutorial from balena like I did with balenaSound but I get an error and I'm not able to install it due to it. Error log: https://pastebin.com/SPNRV4hw

klutchell, 2020-04-06:
Can you confirm what device you're running, and which architecture?

The Dockerfiles in this repo are meant for rpi-3 32-bit and rpi-zero.

Can you try renaming `Dockerfile.armv7hf` to `Dockerfile` in both the pihole and dnscrypt-proxy folders and let me know if that changes anything?

Vysp3r, 2020-04-06:
Changing it from Dockerfile.armv7hf to Dockerfile fixed everything. Tanks a lot for the quick reply too!

closed 2020-04-06 by Vysp3r

klutchell, 2020-04-06:
Can you confirm what device you're running, and which architecture, so I can support it better in the future?