klutchell/balena-pihole #38: Can't build Service

idoodler, 2020-04-24:
I just did some modification to my Balena cloud project which is utilizing your your balena-pihole module. However currently the service `DnscryptProxy` is unable to be built. Here is the log:

```
[DnscryptProxy]  standard_init_linux.go:207: exec user process caused "no such file or directory"
[DnscryptProxy]  
[DnscryptProxy]  Removing intermediate container 2b511a2f05bf
[DnscryptProxy]  The command '/bin/sh -c apk add --no-cache ca-certificates curl        && curl -L "${DNSCRYPT_PROXY_URL}${DNSCRYPT_PROXY_VERSION}.tar.gz" -o /tmp/dnscrypt-proxy.tar.gz        && tar xzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt  && go build -v -ldflags="-s -w"' returned a non-zero code: 1
```

klutchell, 2020-04-24:
What type of device are you building for, and is the application set to 32-bit or 64-bit?

Generally if you build with `balena push` or `git push balena master` it will build for the type of Application you created in the Balena Dashboard.

idoodler, 2020-04-24:
I am building it for Raspberry Pi 3. I don't know what Arch the application is set, I can't find any information about that in the Balena Dashboard or the log.

Isn't the Arch specific to the device part of the application?

klutchell, 2020-04-24:
When creating your application for Raspberry Pi 3, there would have been an optional 64-bit OS type.

[image]

The little raspberry icon next to your application name will either say '3' or '64'.

Can you include the rest of the logs from the failure?

denisinla, 2020-04-24:
I've had the same issues when updating my devices. Looks like the build process does hiccup when running `balena push <deviceName>` since the packages are built on the balena servers then pushed over to the rpi device. Try running the command once more.

idoodler, 2020-04-24:
@klutchell The device is of type 3.

@denisinla I can reproduce the same behaviour every time, even on different build nodes. I have seen different nodes being used, from arm01 to arm03.

denisinla, 2020-04-24:
I did some more digging and looks like https://github.com/klutchell/dnscrypt-proxy/commit/cc1cfa5c18bd4bc45270a8804962cd180addce82 has a `dnscrypt` URL that is 404'ing.

klutchell, 2020-04-24:
@denisinla I need to see the top of the balena build logs so I know which image it's using as a base. I suspect it's using the wrong architecture and I have a workaround in mind but I need to confirm with the full log including the base image pull.

idoodler, 2020-04-24:
@klutchell Here you go:
<details>
  <summary>Full Log</summary>
  
```
 [Info]           Starting build for xxxxx, user xxxxx
[Info]           Dashboard link: https://dashboard.balena-cloud.com/apps/xxxxx/devices
[Info]           Building on arm01
[Info]           Pulling previous images for caching purposes...
[Success]        Successfully pulled cache images
[PiHole]         Step 1/17 : FROM balenalib/raspberrypi3-debian:jessie-build AS build
[DnscryptProxy]  Step 1/17 : FROM balenalib/armv7hf-alpine-golang:1.13 as build
[PiHole]          ---> 2da9d4d027c0
[PiHole]         Step 2/17 : WORKDIR /usr/src
[PiHole]         Using cache
[PiHole]          ---> 13e1d014d621
[PiHole]         Step 3/17 : RUN install_packages cmake libraspberrypi-dev
[PiHole]         Using cache
[PiHole]          ---> dabecf57ae8f
[PiHole]         Step 4/17 : RUN curl -sSL https://raw.githubusercontent.com/tasanakorn/rpi-fbcp/master/CMakeLists.txt -O
[PiHole]         Using cache
[PiHole]          ---> 11b9ae9b4a5b
[PiHole]         Step 5/17 : RUN curl -sSL https://raw.githubusercontent.com/tasanakorn/rpi-fbcp/master/main.c -O
[PiHole]         Using cache
[PiHole]          ---> 96a87e04207c
[PiHole]         Step 6/17 : WORKDIR /usr/src/build
[PiHole]         Using cache
[PiHole]          ---> 0d37d9f3040c
[PiHole]         Step 7/17 : RUN cmake .. && make
[PiHole]         Using cache
[PiHole]          ---> a136b0dbc79f
[PiHole]         Step 8/17 : FROM pihole/pihole:v4.4-armhf
[DnscryptProxy]   ---> bbe171f91a3c
[DnscryptProxy]  Step 2/17 : WORKDIR /go/src/github.com/DNSCrypt/dnscrypt-proxy/
[DnscryptProxy]  Using cache
[DnscryptProxy]   ---> 87fd58cc624a
[DnscryptProxy]  Step 3/17 : ARG DNSCRYPT_PROXY_VERSION=2.0.39
[DnscryptProxy]  Using cache
[DnscryptProxy]   ---> 62601c15e8e5
[DnscryptProxy]  Step 4/17 : ARG DNSCRYPT_PROXY_URL=https://github.com/DNSCrypt/dnscrypt-proxy/archive/
[PiHole]          ---> 96082728873a
[PiHole]         Step 9/17 : WORKDIR /usr/src
[DnscryptProxy]  Using cache
[DnscryptProxy]   ---> 8aeaf0a5e3eb
[DnscryptProxy]  Step 5/17 : ENV CGO_ENABLED 0
[DnscryptProxy]  Using cache
[DnscryptProxy]   ---> 400568d5951e
[DnscryptProxy]  Step 6/17 : RUN apk add --no-cache ca-certificates curl        && curl -L "${DNSCRYPT_PROXY_URL}${DNSCRYPT_PROXY_VERSION}.tar.gz" -o /tmp/dnscrypt-proxy.tar.gz        && tar xzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt  && go build -v -ldflags="-s -w"
[PiHole]         Using cache
[PiHole]          ---> cc483a65a0ea
[PiHole]         Step 10/17 : COPY --from=build /opt/vc/lib/* /opt/vc/lib/
[DnscryptProxy]   ---> Running in f909c5173567
[PiHole]         Using cache
[PiHole]          ---> 40abc303b458
[PiHole]         Step 11/17 : COPY --from=build /usr/src/build/fbcp /usr/src/
[PiHole]         Using cache
[PiHole]          ---> fbf03fc34bd0
[PiHole]         Step 12/17 : COPY services/ /etc/services.d/
[PiHole]         Using cache
[PiHole]          ---> 95e1fc25e30a
[PiHole]         Step 13/17 : COPY init/ /etc/cont-init.d/
[PiHole]         Using cache
[PiHole]          ---> 2701d039891b
[PiHole]         Step 14/17 : RUN sed -i '/$AUTHORIZED_HOSTNAMES = array(/ a "balena-devices.com",' /var/www/html/admin/scripts/pi-hole/php/auth.php
[PiHole]         Using cache
[PiHole]          ---> f6625000f43a
[PiHole]         Step 15/17 : RUN echo "bind-interfaces" >> /etc/dnsmasq.conf
[PiHole]         Using cache
[PiHole]          ---> f13f4b4e0542
[PiHole]         Step 16/17 : SHELL ["/bin/bash", "-o", "pipefail", "-c"]
[PiHole]         Using cache
[PiHole]          ---> 0cd547ac1e7c
[PiHole]         Step 17/17 : RUN curl -fsSL https://github.com/jpmck/PADD/archive/v3.1.tar.gz | tar xz --strip-components 1    && chmod +x padd.sh
[PiHole]         Using cache
[PiHole]          ---> 6ad9cc4f1817
[PiHole]         Successfully built 6ad9cc4f1817
[DnscryptProxy]  standard_init_linux.go:207: exec user process caused "no such file or directory"
[DnscryptProxy]  
[DnscryptProxy]  Removing intermediate container f909c5173567
[DnscryptProxy]  The command '/bin/sh -c apk add --no-cache ca-certificates curl        && curl -L "${DNSCRYPT_PROXY_URL}${DNSCRYPT_PROXY_VERSION}.tar.gz" -o /tmp/dnscrypt-proxy.tar.gz        && tar xzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt  && go build -v -ldflags="-s -w"' returned a non-zero code: 1
[Info]           Uploading images
[Success]        Successfully uploaded images
[Error]          Some services failed to build:
[Error]            Service: DnscryptProxy
[Error]              Error: The command '/bin/sh -c apk add --no-cache ca-certificates curl     && curl -L "${DNSCRYPT_PROXY_URL}${DNSCRYPT_PROXY_VERSION}.tar.gz" -o /tmp/dnscrypt-proxy.tar.gz        && tar xzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt  && go build -v -ldflags="-s -w"' returned a non-zero code: 1
[Info]           Built on arm01
[Error]          Not deploying release.
Remote build failed
```
</details>

I removed all other serviced to reduce the log length

klutchell, 2020-04-24:
Thank you for the logs.

Could you please both try changing the 1st `FROM` line of `dnscrypt-proxy/Dockerfile.armv7hf`?

Before:
```Dockerfile
FROM balenalib/armv7hf-alpine-golang:1.13 as build
```
After:
```Dockerfile
FROM balenalib/raspberrypi3-alpine-golang:1.13 as build
```

And let me know if it builds successfully? If it works for both of you I'll push the change and investigate the root cause over the weekend.

denisinla, 2020-04-24:
Didn't work on my end. Also forgot it was Friday lels.

denisinla, 2020-04-24:
Looks like `balenalib/raspberrypi3-alpine-golang:1.13` is not available only `1.12.9` is. Will try with that version or `latest`.

denisinla, 2020-04-24:
Failed again with the same error for dnscrypt-proxy.

klutchell, 2020-04-24:
Good catch, but it actually seems to impact all of the `balenalib/raspberrypi3-alpine-golang` and `balenalib/armv7hf-alpine-golang` images. I just pushed a fix by switching to the non-alpine variant to build the base.

idoodler, 2020-04-25:
@klutchell I can confirm that your latest push fixes my issue

closed 2020-04-25 by idoodler