klutchell/balena-pihole #39: 'Dockerfile parse error' pushing Pi-Hole to RPi4

MacularDegenerate, 2020-04-26:
I’m trying to run pi-hole on a Raspberry Pi 4B, but when I attempt to push the project code to my device, I always get the following error message. I am deploying the code from a Windows machine and made sure to push with --convert-eol.

`[Error]           Some services failed to build:
[Error]             Service: pihole
[Error]               Error: Dockerfile parse error line 1: unknown instruction: DOCKERFILE.ARMV7HF`

Has anyone else run into this? Any suggestions? Thanks in advance!

ddimm, 2020-04-26:
I was also getting the same error on Windows. What worked for me was copying the contents of `DOCKERFILE.ARMV7HF` into `Dockerfile.aarch64` in the `pihole` directory.  This resulted in a different error which was fixed by replacing `FROM balenalib/raspberrypi3-debian:jessie-build AS build` with `FROM balenalib/rpi-debian:jessie-build AS build`. Hope this helps!

klutchell, 2020-04-26:
I had forgotten that symlinks don't work with WIndows workspaces. I've pushed the fix now.

MacularDegenerate, 2020-04-26:
This seems to have solved the problem! Thanks for the prompt response.

closed 2020-04-27 by klutchell