klutchell/balena-pihole #43: PADD not showing stats in 5.1.1

eiddor, 2020-07-19:
I just pushed the 5.1.1 release, and though everything loads and displays properly, PADD isn't showing any pihole stats (it does show environment stats of the RPi).

I manually updated PADD in the pihole container and am still having the issue.

klutchell, 2020-07-22:
I don't currently have my screen connected, but can you try installing the latest PADD release via the Dockerfile instead of running the upgrader? I want to make sure your changes don't get lost on restart.

Just change the following line:
https://github.com/klutchell/balena-pihole/blob/master/pihole/Dockerfile#L31

```
RUN curl -fsSL https://github.com/pi-hole/PADD/archive/v3.2.2.tar.gz | tar xz --strip-components 1 \
	&& chmod +x padd.sh
```

eiddor, 2020-07-22:
Hey Kyle - Same results.  The newer padd.sh is downloaded (I did a "Purge Data" to be sure), but I'm not seeing any pihole stats.

UPDATE: I just checked the Pi-hole/PADD repo, and this is due to an FTL change in Pi-hole.  The workaround is to type this in the pihole container: 

`echo 4711 | sudo tee /var/run/pihole-FTL.port`

It's working now and will be updated in the next Pi-hole release.

Sorry for the fire drill!

klutchell, 2020-07-22:
Thanks for investigating! It looks like PADD is nearing EOL from the original developer? Not sure how it will be supported going forward.

Do you have a link to the documentation where you found the workaround? I can add a step to the Dockerfile if it's a known issue with FTL 5.1 and will help others.

eiddor, 2020-07-22:
Sure thing!  It's covered in this issue post (there are others as well):

https://github.com/pi-hole/PADD/issues/112#issuecomment-660264727

klutchell, 2020-07-22:
Thanks! It looks like they are planning a fix in FTL so I'm not going to change this repo for now. If others encounter the same issue I'll direct them to that workaround.

closed 2020-07-22 by klutchell