klutchell/balena-pihole #44: Building error

clawfire, 2020-08-30:
When using the 1-click button to copy the app into my balena cloud account, I got this for the release build: 

```
[Info]            Starting build for balena-pihole, user gh_clawfire
[Info]            Dashboard link: https://dashboard.balena-cloud.com/apps/1730930/devices
[Info]            Building on arm03
[Info]            Pulling previous images for caching purposes...
[Success]         Successfully pulled cache images
[Info]            Starting to build dnscrypt-proxy
[Info]            Starting to build pihole
[Success]         Successfully built service pihole
[Info]            Uploading images
[Success]         Successfully uploaded images
[Error]           Some services failed to build:
[Error]             Service: dnscrypt-proxy
[Error]               Error: The command '/bin/sh -c install_packages ca-certificates curl     && curl -L "${DNSCRYPT_PROXY_URL}${DNSCRYPT_PROXY_VERSION}.tar.gz" -o /tmp/dnscrypt-proxy.tar.gz      && tar xzf /tmp/dnscrypt-proxy.tar.gz --strip 1 -C /go/src/github.com/DNSCrypt        && go build -v -ldflags="-s -w"' returned a non-zero code: 1
[Info]            Built on arm03
[Error]           Not deploying release.
```

klutchell, 2020-08-31:
It looks like there was a temporary error in downloading the latest dnscrypt-proxy release from GitHub. Possibly related to the Cloudflare outages yesterday?

Either way, if you try deploying again today I expect it will succeed as the sources haven't changed recently.

clawfire, 2020-08-31:
Well, tried a fresh build 11 minutes ago and... still have the same issue :/

clawfire, 2020-08-31:
where `DNSCRYPT_PROXY_URL` and `DNSCRYPT_PROXY_VERSION` are supposed to be defined?

klutchell, 2020-08-31:
They are defined in the top of the Dockerfile, at lines 5 & 6.

https://github.com/klutchell/balena-pihole/blob/master/dnscrypt-proxy/Dockerfile.template#L5-L6

I just tried a build locally and it worked for me. What type of device are you building for? Do you have any additional logs or just what you posted above?

clawfire, 2020-09-07:
I'm using the 1-click button deploy in balena cloud platform. Gonna try the local build

klutchell, 2020-09-07:
I also tried the one-click method and it worked for me. What default device type did you select for your application in the Balena dashboard?

clawfire, 2020-09-07:
pi-zero

klutchell, 2020-09-07:
Unfortunately one-click deploy will only work for rpi3 and rpi4 with this repo. This is because for rpi-zero we have to select a different upstream pihole image that doesn't follow balena's naming convention.

However if you download the repo, and rename `Dockerfile.rpi` to `Dockerfile` (overwrite the existing one) and push via balena cli or via git it should work for rpi-zero.

Or a 3rd option, you could fork the repo and in your fork perform the rename outlined above. Then one-click-deploy would work from your repo!

clawfire, 2020-09-08:
Did it, the building is fixed 😃 
Ran into other issues but that's another Issue 😃 
Thanks

closed 2020-09-08 by clawfire