klutchell/balena-pihole #45: Processes don't use the config file

clawfire, 2020-09-08:
After solving #44 , I ran into another issue which is, despite I changed the `balena.yml` file to use `wlan0` I can see the both `pihole` process and `dnscript-proxy` using `eth0` which my pi zero don't have. 

```
08.09.20 20:21:12 (+0200) Installing service 'dnscrypt-proxy sha256:786c5d65a7e0ad8c4c655f3f678281b5048315cdf522e0d87e0aa00e8c50b391'
08.09.20 20:21:12 (+0200) Installing service 'pihole sha256:9653e07ef791894c92b58eb4cc78c3aadf0b3ca8f84980c6eaa82a4d27434dbd'
08.09.20 20:21:14 (+0200) Installed service 'dnscrypt-proxy sha256:786c5d65a7e0ad8c4c655f3f678281b5048315cdf522e0d87e0aa00e8c50b391'
08.09.20 20:21:14 (+0200) Starting service 'dnscrypt-proxy sha256:786c5d65a7e0ad8c4c655f3f678281b5048315cdf522e0d87e0aa00e8c50b391'
08.09.20 20:21:14 (+0200) Installed service 'pihole sha256:9653e07ef791894c92b58eb4cc78c3aadf0b3ca8f84980c6eaa82a4d27434dbd'
08.09.20 20:21:14 (+0200) Starting service 'pihole sha256:9653e07ef791894c92b58eb4cc78c3aadf0b3ca8f84980c6eaa82a4d27434dbd'
08.09.20 20:21:28 (+0200) Started service 'pihole sha256:9653e07ef791894c92b58eb4cc78c3aadf0b3ca8f84980c6eaa82a4d27434dbd'
08.09.20 20:21:26 (+0200)  pihole  [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
08.09.20 20:21:28 (+0200)  pihole  [s6-init] ensuring user provided files have correct perms...exited 0.
08.09.20 20:21:28 (+0200)  pihole  [fix-attrs.d] applying ownership & permissions fixes...
08.09.20 20:21:28 (+0200)  pihole  [fix-attrs.d] 01-resolver-resolv: applying... 
08.09.20 20:21:28 (+0200)  pihole  [fix-attrs.d] 01-resolver-resolv: exited 0.
08.09.20 20:21:28 (+0200)  pihole  [fix-attrs.d] done.
08.09.20 20:21:28 (+0200)  pihole  [cont-init.d] executing container initialization scripts...
08.09.20 20:21:28 (+0200)  pihole  [cont-init.d] 20-start.sh: executing... 
08.09.20 20:21:30 (+0200)  pihole   ::: Starting docker specific checks & setup for docker pihole/pihole
08.09.20 20:21:30 (+0200)  pihole  OK: Checks passed for /etc/resolv.conf DNS servers
08.09.20 20:21:30 (+0200)  pihole  
08.09.20 20:21:30 (+0200)  pihole  nameserver 127.0.0.1
08.09.20 20:21:30 (+0200)  pihole  nameserver 1.1.1.1
08.09.20 20:21:30 (+0200)  pihole  options timeout:15
08.09.20 20:21:30 (+0200)  pihole  Assigning random password: -mdrNUeQ
08.09.20 20:21:33 (+0200)  pihole    [i] Existing PHP installation detected : PHP version 7.0.33-0+deb9u7
08.09.20 20:21:34 (+0200) Started service 'dnscrypt-proxy sha256:786c5d65a7e0ad8c4c655f3f678281b5048315cdf522e0d87e0aa00e8c50b391'
08.09.20 20:21:32 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:32] [NOTICE] dnscrypt-proxy 2.0.44
08.09.20 20:21:32 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:32] [NOTICE] Network connectivity detected
08.09.20 20:21:32 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:32] [NOTICE] Now listening to 0.0.0.0:5053 [UDP]
08.09.20 20:21:32 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:32] [NOTICE] Now listening to 0.0.0.0:5053 [TCP]
08.09.20 20:21:32 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:32] [NOTICE] Source [public-resolvers] loaded
08.09.20 20:21:33 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:33] [NOTICE] Source [relays] loaded
08.09.20 20:21:33 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:33] [NOTICE] Firefox workaround initialized
08.09.20 20:21:33 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:33] [NOTICE] [deffer-dns.au] OK (DNSCrypt) - rtt: 317ms
08.09.20 20:21:36 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:36] [NOTICE] [powerdns-doh] OK (DoH) - rtt: 106ms
08.09.20 20:21:36 (+0200)  pihole  
08.09.20 20:21:36 (+0200)  pihole    [i] Installing configs from /etc/.pihole...
08.09.20 20:21:36 (+0200)  pihole    [i] Existing dnsmasq.conf found... it is not a Pi-hole file, leaving alone!
08.09.20 20:21:36 (+0200)  pihole    [i] Copying 01-pihole.conf to /etc/dnsmasq  [✓] Copying 01-pihole.conf to /etc/dnsmasq.d/01-pihole.conf
08.09.20 20:21:37 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:37] [NOTICE] [dnslify-doh] OK (DoH) - rtt: 226ms
08.09.20 20:21:37 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:37] [NOTICE] [dnscrypt.be] OK (DNSCrypt) - rtt: 53ms
08.09.20 20:21:37 (+0200)  pihole  chown: cannot access '/etc/pihole/dhcp.leases': No such file or directory
08.09.20 20:21:38 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:38] [NOTICE] [dnscrypt.ca-1-doh] OK (DoH) - rtt: 157ms
08.09.20 20:21:38 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:38] [NOTICE] [v.dnscrypt.uk-ipv4] OK (DNSCrypt) - rtt: 60ms
08.09.20 20:21:39 (+0200)  pihole  ::: Pre existing WEBPASSWORD found
08.09.20 20:21:39 (+0200)  pihole  Using default DNS servers: 8.8.8.8 & 8.8.4.4
08.09.20 20:21:39 (+0200)  pihole  DNSMasq binding to default interface: eth0
08.09.20 20:21:40 (+0200)  pihole  Added ENV to php:
08.09.20 20:21:40 (+0200)  pihole                       "PHP_ERROR_LOG" => "/var/log/lighttpd/error.log",
08.09.20 20:21:40 (+0200)  pihole                       "ServerIP" => "0.0.0.0",
08.09.20 20:21:40 (+0200)  pihole                       "VIRTUAL_HOST" => "0.0.0.0",
08.09.20 20:21:40 (+0200)  pihole  Using IPv4 and IPv6
08.09.20 20:21:40 (+0200)  pihole  ::: Preexisting ad list /etc/pihole/adlists.list detected ((exiting setup_blocklists early))
08.09.20 20:21:40 (+0200)  pihole  https://raw.githubusercontent.com/StevenBlack/hosts/master/hosts
08.09.20 20:21:40 (+0200)  pihole  https://mirror1.malwaredomains.com/files/justdomains
08.09.20 20:21:40 (+0200)  pihole  http://sysctl.org/cameleon/hosts
08.09.20 20:21:40 (+0200)  pihole  https://s3.amazonaws.com/lists.disconnect.me/simple_tracking.txt
08.09.20 20:21:40 (+0200)  pihole  https://s3.amazonaws.com/lists.disconnect.me/simple_ad.txt
08.09.20 20:21:40 (+0200)  pihole  https://hosts-file.net/ad_servers.txt
08.09.20 20:21:41 (+0200)  pihole  ::: Testing pihole-FTL DNS: sudo: unable to resolve host 9f5acc1
08.09.20 20:21:42 (+0200)  pihole  
08.09.20 20:21:42 (+0200)  pihole  dnsmasq: unknown interface eth0
08.09.20 20:21:42 (+0200)  pihole  [cont-init.d] 20-start.sh: exited 1.
08.09.20 20:21:42 (+0200)  pihole  [cont-finish.d] executing container finish scripts...
08.09.20 20:21:42 (+0200)  pihole  [cont-finish.d] done.
08.09.20 20:21:42 (+0200)  pihole  [s6-finish] waiting for services.
08.09.20 20:21:42 (+0200)  pihole  [s6-finish] sending all processes the TERM signal.
08.09.20 20:21:46 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:46] [NOTICE] [doh-ibksturm] OK (DoH) - rtt: 66ms
08.09.20 20:21:46 (+0200)  pihole  [s6-finish] sending all processes the KILL signal and exiting.
[... 4 lines cut ...]
08.09.20 20:21:53 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:53] [NOTICE] [faelix-uk-ipv4] OK (DNSCrypt) - rtt: 56ms
08.09.20 20:21:54 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:54] [NOTICE] [bortzmeyer] OK (DoH) - rtt: 36ms
08.09.20 20:21:54 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:54] [NOTICE] [acsacsar-ams-ipv4] OK (DNSCrypt) - rtt: 35ms
08.09.20 20:21:54 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:54] [NOTICE] [dns.digitale-gesellschaft.ch] OK (DoH) - rtt: 68ms
08.09.20 20:21:54 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:54] [NOTICE] [sth-dnscrypt-se] OK (DNSCrypt) - rtt: 126ms
08.09.20 20:21:55 (+0200) Service exited 'pihole sha256:9653e07ef791894c92b58eb4cc78c3aadf0b3ca8f84980c6eaa82a4d27434dbd'
08.09.20 20:21:55 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:55] [NOTICE] [dnscrypt.ca-2-doh] OK (DoH) - rtt: 129ms
08.09.20 20:21:56 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:56] [NOTICE] [faelix-ch-ipv4-doh] OK (DoH) - rtt: 48ms
08.09.20 20:21:56 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:56] [NOTICE] [jp.tiarap.org] OK (DoH) - rtt: 46ms
08.09.20 20:21:57 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:57] [NOTICE] [ev-va] OK (DNSCrypt) - rtt: 181ms
08.09.20 20:21:59 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:59] [NOTICE] [bcn-doh] OK (DoH) - rtt: 99ms
08.09.20 20:21:59 (+0200)  dnscrypt-proxy  [2020-09-08 18:21:59] [NOTICE] [doh.ffmuc.net] OK (DoH) - rtt: 73ms
08.09.20 20:22:00 (+0200)  dnscrypt-proxy  [2020-09-08 18:22:00] [NOTICE] [cz.nic] OK (DoH) - rtt: 65ms
08.09.20 20:22:01 (+0200)  dnscrypt-proxy  [2020-09-08 18:22:01] [NOTICE] [nextdns] OK (DoH) - rtt: 47ms
08.09.20 20:22:04 (+0200)  dnscrypt-proxy  [2020-09-08 18:22:04] [NOTICE] [doh.appliedprivacy.net] OK (DoH) - rtt: 97ms
08.09.20 20:22:04 (+0200)  dnscrypt-proxy  [2020-09-08 18:22:04] [NOTICE] [meganerd] OK (DNSCrypt) - rtt: 58ms
08.09.20 20:22:04 (+0200)  dnscrypt-proxy  [2020-09-08 18:22:04] [NOTICE] [soltysiak] OK (DNSCrypt) - rtt: 67ms
[... cut ...]
08.09.20 20:22:10 (+0200)  dnscrypt-proxy  [2020-09-08 18:22:10] [NOTICE] [publicarray-au2] OK (DNSCrypt) - rtt: 293ms
08.09.20 20:22:10 (+0200)  dnscrypt-proxy  [2020-09-08 18:22:10] [NOTICE] [jp.tiar.app] OK (DNSCrypt) - rtt: 285ms
08.09.20 20:22:10 (+0200)  dnscrypt-proxy  [2020-09-08 18:22:10] [NOTICE] [faelix-uk-ipv4-doh] OK (DoH) - rtt: 95ms
08.09.20 20:22:12 (+0200)  dnscrypt-proxy  [2020-09-08 18:22:12] [NOTICE] [publicarray-au-doh] OK (DoH) - rtt: 295ms
08.09.20 20:22:12 (+0200) Restarting service 'pihole sha256:9653e07ef791894c92b58eb4cc78c3aadf0b3ca8f84980c6eaa82a4d27434dbd'
```

klutchell, 2020-09-08:
`balena.yml` is only used with one-click-deploy. If you deployed via any other method, like via git push or balena cli, you'll need to set those environment variables manually in your application dashboard. Also if you pushed once via one-click-deploy and then updated your application and pushed again, the environment variables will not be changed.

Go to your app dashboard and select Environment Variables in the leftmost column to update those values.

clawfire, 2020-09-08:
Oh! I missed that part. Feeling a bit dumb 😄  Thanks for the heads up. gonna check that.

clawfire, 2020-09-08:
And it works like a charm now. I had cached data from the 1click deploy. Thanks

closed 2020-09-08 by clawfire