klutchell/balena-pihole #46: dnsmasq and port 53 conflict: bind-interfaces

borgmon, 2020-09-27:
I see the code is trying to [fix ](https://github.com/klutchell/balena-pihole/blob/6f5cb45628570036f76fad64513b54a9236088e7/pihole/Dockerfile#L28)the `dnsmasq` issue by adding `bind-interfaces`.
However, there is another way mentioned by docker-pi-hole [here](https://github.com/pi-hole/docker-pi-hole/issues/491). Should we use it instead?

References:
- https://www.balena.io/docs/learn/develop/runtime/#using-dns-resolvers-in-your-container
- https://github.com/pi-hole/docker-pi-hole/issues/491

klutchell, 2020-09-28:
The issue you linked is still open, and in the comments they are using the same `bind-interfaces` solution that we are.

https://github.com/pi-hole/docker-pi-hole/issues/491

The environment variable `DNSMASQ_BIND_INTERFACES` as recommended by @richardpowellus does not exist yet and it was only a suggestion for a new feature.

Unless you saw another solution in this thread that I missed?

klutchell, 2020-10-13:
I setup notifications on the thread you linked. If they add support for `DNSMASQ_BIND_INTERFACES` then we will switch to that method. For now we are stuck with the `bind-interfaces` flag.

closed 2020-10-13 by klutchell