klutchell/balena-pihole #50: Duplicate padd.sh processes preventing startup

eiddor, 2020-11-30:
This issue has been plaguing me for a long time, but I've always just worked-around it thinking that it was local.

Following a reboot, restart, or new push (restart), my Pi-hole/padd (RPi3 with TFT display) gets stuck at this stage which is displayed on the screen:

```
- Writing PID (1135) to file...
- Checking for PROD version file...
   - PROD version file not found.
- Gathering system information...
- Gathering Pi-hole information...
```

If I look at the pihole container, I can see two padd.sh processes trying to start 

```
/usr/src# ps -ef | grep padd
root      1114     1  0 08:18 pts/0    00:00:00 s6-supervise padd
root      1135  1119  8 08:18 ?        00:02:19 bash /usr/src/padd.sh
root      1293  1135 99 08:18 ?        00:26:49 bash /usr/src/padd.sh
```

Killing the second PID (1293 in this case) allows things to continue and the pihole fully starts.  Judging by the PIDs, it looks like the first process (1135) is spawning the second, but I'm not comfortable enough in shell scripting to figure out where the problem lies (if there is a problem).

I've tried starting with a fresh machine and though it works the first time, subsequent pushes will trigger this issue.

I'm running the latest version as of today, but this issue has been present for me for over a year.

My RPi3 is running balenaOS 2.58.3+rev1 / Sup 11.14.0.

Let me know if I can provide any other information.

Thanks!
Roddie

klutchell, 2020-11-30:
Hmm, interesting find. I think I've seen the same behaviour in my alternate setup and I worked around it by adding a delay on PADD startup. At the time I hadn't noticed two processes though so that's helpful!

https://github.com/klutchell/balena-pihole-unbound/blob/14689cf8feb8be6d85664695abffd5bcd25a6934/pihole/services/padd/run#L5

Obviously this isn't a proper fix but it might be worthwhile to port the change to this repo if others are experiencing it. At least until we can figure out something better.

eiddor, 2020-11-30:
Thanks, Kyle!  That worked for me across a push and across a reboot.

closed 2020-12-01 by klutchell