klutchell/balena-pihole #55: Replace fbcp service with balenablocks/fbcp

klutchell, 2021-01-04:
There is a balenaBlocks image that we should pull and avoid forcing fbcp into the pihole image.

https://github.com/balenablocks/fbcp

klutchell, 2021-01-04:
cc @rahul-thakoor in case he wants to assist

rahul-thakoor, 2021-01-04:
I can help with that @klutchell !

toastyghost710, 2021-01-20:
> 20.01.21 11:02:46 (-0600)  fbcp  Display not detected, exiting...
20.01.21 11:02:46 (-0600)  fbcp  Display not detected, exiting...
20.01.21 11:02:46 (-0600)  pihole  [s6-finish] sending all processes the KILL signal and exiting.
20.01.21 11:02:47 (-0600) Service exited 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'
20.01.21 11:02:48 (-0600) Service exited 'pihole sha256:bc413244e6892e904f1255585a45a179e695184e7a8e27a562a0088b6f66295b'
20.01.21 11:02:48 (-0600) Killed service 'pihole sha256:bc413244e6892e904f1255585a45a179e695184e7a8e27a562a0088b6f66295b'

@klutchell it appears without an external screen that the pihole container fails to start?

klutchell, 2021-01-20:
Hey @toastyghost710 can you open a new issue and include more logs (above and below what you've included here) and your OS version? For extra detail you can open a HostOS shell and run `cat /etc/os-release'.

closed 2021-01-21 by klutchell