klutchell/balena-pihole #57: Recent update to unbound.conf seems to be causing DNS errors

toastyghost710, 2021-01-17:
Upon implementing the most recent update including the updated unbound.conf to my Balena Cloud, I immediately lose internet access and am greeted with a DNS_PROBE_FINISHED_BAD_CONFIG error with any attempt to load a webpage.

klutchell, 2021-01-17:
Yeah, I was worried this might happen. For some dumb reason I changed the default Unbound port from `5053` to `5335` so that would require changing your `DNS1` and `DNS2` env vars to match.

I changed the port back to `5053` now though, so feel free to grab the latest commit.

https://github.com/klutchell/balena-pihole/commit/910a0eec79421eab6fc1a3aa326f023174e31446

klutchell, 2021-01-20:
@toastyghost710 is the latest commit working for you?

toastyghost710, 2021-01-20:
@klutchell yes it is. My apologies, I meant to write back that day to inform you that the changes fixed the issues.

klutchell, 2021-01-20:
Glad to hear it, take care!

closed 2021-01-20 by klutchell