klutchell/balena-pihole #58: Container restarting endlessly (Pi Zero)

caiorg, 2021-01-19:
This is, apparently, the exact same issue mentioned at #15, but opening a new one as requested by @klutchell.

The issue is I've tried to update PiHole version from 4.4 (installed by the One Click deploy button) to 5.4 (or any subversion from major version 5), since I know my PiZero can run it (installed it manually using another tutorial), but when the container tries to start, it keeps restarting.

Also on PiHole's version 4.4 (using balena-pihole), I'm stuck with the "Domains in blocklist" card showing -2.

Running host os version: balenaOS 2.54.2+rev1

@klutchell support access can be granted, if needed.

This is just a snippet (full log can be provided)
```
19.01.21 20:30:10 (-0300) Starting service 'pihole sha256:7bdea0d0f0b62420fea0e71c20d8eebd37a4fef8dc6f3ae5d85731f16c400843'
19.01.21 20:30:18 (-0300) Started service 'pihole sha256:7bdea0d0f0b62420fea0e71c20d8eebd37a4fef8dc6f3ae5d85731f16c400843'
19.01.21 20:29:12 (-0300)  pihole  [s6-finish] sending all processes the KILL signal and exiting.
19.01.21 20:30:17 (-0300)  pihole  [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
19.01.21 20:30:19 (-0300)  pihole  [s6-init] ensuring user provided files have correct perms...exited 0.
19.01.21 20:30:19 (-0300)  pihole  [fix-attrs.d] applying ownership & permissions fixes...
19.01.21 20:30:19 (-0300)  pihole  [fix-attrs.d] 01-resolver-resolv: applying... 
19.01.21 20:30:19 (-0300)  pihole  [fix-attrs.d] 01-resolver-resolv: exited 0.
19.01.21 20:30:19 (-0300)  pihole  [fix-attrs.d] done.
19.01.21 20:30:19 (-0300)  pihole  [cont-init.d] executing container initialization scripts...
19.01.21 20:30:19 (-0300)  pihole  [cont-init.d] 20-start.sh: executing... 
19.01.21 20:30:19 (-0300)  pihole  [cont-init.d] 20-start.sh: exited 267.
19.01.21 20:30:19 (-0300)  pihole  [cont-finish.d] executing container finish scripts...
19.01.21 20:30:19 (-0300)  pihole  [cont-finish.d] done.
19.01.21 20:30:19 (-0300)  pihole  [s6-finish] waiting for services.
19.01.21 20:30:20 (-0300)  pihole  [s6-finish] sending all processes the TERM signal.
19.01.21 20:30:23 (-0300)  pihole  [s6-finish] sending all processes the KILL signal and exiting.
19.01.21 20:30:27 (-0300) Service exited 'pihole sha256:7bdea0d0f0b62420fea0e71c20d8eebd37a4fef8dc6f3ae5d85731f16c400843'
19.01.21 20:30:34 (-0300) Restarting service 'pihole sha256:7bdea0d0f0b62420fea0e71c20d8eebd37a4fef8dc6f3ae5d85731f16c400843'
19.01.21 20:30:23 (-0300)  pihole  [s6-finish] sending all processes the KILL signal and exiting.
19.01.21 20:30:34 (-0300)  pihole  [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
19.01.21 20:30:35 (-0300)  pihole  [s6-init] ensuring user provided files have correct perms...exited 0.
19.01.21 20:30:35 (-0300)  pihole  [fix-attrs.d] applying ownership & permissions fixes...
19.01.21 20:30:35 (-0300)  pihole  [fix-attrs.d] 01-resolver-resolv: applying... 
19.01.21 20:30:35 (-0300)  pihole  [fix-attrs.d] 01-resolver-resolv: exited 0.
19.01.21 20:30:35 (-0300)  pihole  [fix-attrs.d] done.
19.01.21 20:30:35 (-0300)  pihole  [cont-init.d] executing container initialization scripts...
19.01.21 20:30:35 (-0300)  pihole  [cont-init.d] 20-start.sh: executing... 
19.01.21 20:30:35 (-0300)  pihole  [cont-init.d] 20-start.sh: exited 267.
19.01.21 20:30:35 (-0300)  pihole  [cont-finish.d] executing container finish scripts...
19.01.21 20:30:35 (-0300)  pihole  [cont-finish.d] done.
19.01.21 20:30:35 (-0300)  pihole  [s6-finish] waiting for services.
19.01.21 20:30:36 (-0300)  pihole  [s6-finish] sending all processes the TERM signal.
19.01.21 20:30:39 (-0300)  pihole  [s6-finish] sending all processes the KILL signal and exiting.
19.01.21 20:30:43 (-0300) Service exited 'pihole sha256:7bdea0d0f0b62420fea0e71c20d8eebd37a4fef8dc6f3ae5d85731f16c400843'
19.01.21 20:30:51 (-0300) Killing service 'pihole sha256:7bdea0d0f0b62420fea0e71c20d8eebd37a4fef8dc6f3ae5d85731f16c400843'
19.01.21 20:30:51 (-0300) Restarting service 'pihole sha256:7bdea0d0f0b62420fea0e71c20d8eebd37a4fef8dc6f3ae5d85731f16c400843'
19.01.21 20:30:39 (-0300)  pihole  [s6-finish] sending all processes the KILL signal and exiting.
19.01.21 20:30:50 (-0300)  pihole  [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
19.01.21 20:30:52 (-0300)  pihole  [s6-init] ensuring user provided files have correct perms...exited 0.
19.01.21 20:30:52 (-0300)  pihole  [fix-attrs.d] applying ownership & permissions fixes...
19.01.21 20:30:52 (-0300)  pihole  [fix-attrs.d] 01-resolver-resolv: applying... 
19.01.21 20:30:52 (-0300)  pihole  [fix-attrs.d] 01-resolver-resolv: exited 0.
19.01.21 20:30:53 (-0300)  pihole  [fix-attrs.d] done.
19.01.21 20:30:53 (-0300)  pihole  [cont-init.d] executing container initialization scripts...
19.01.21 20:30:53 (-0300)  pihole  [cont-init.d] 20-start.sh: executing... 
19.01.21 20:30:53 (-0300)  pihole  [cont-init.d] 20-start.sh: exited 267.
19.01.21 20:30:53 (-0300)  pihole  [cont-finish.d] executing container finish scripts...
19.01.21 20:30:53 (-0300)  pihole  [cont-finish.d] done.
19.01.21 20:30:53 (-0300)  pihole  [s6-finish] waiting for services.
19.01.21 20:30:53 (-0300)  pihole  [s6-finish] sending all processes the TERM signal.
19.01.21 20:30:56 (-0300)  pihole  [s6-finish] sending all processes the KILL signal and exiting.
19.01.21 20:31:01 (-0300) Killed service 'pihole sha256:7bdea0d0f0b62420fea0e71c20d8eebd37a4fef8dc6f3ae5d85731f16c400843'
19.01.21 20:31:01 (-0300) Service exited 'pihole sha256:7bdea0d0f0b62420fea0e71c20d8eebd37a4fef8dc6f3ae5d85731f16c400843'
```

Any help will be very much appreciated! 😄

klutchell, 2021-01-20:
Hey @caiorg!

The `-2 Domains in blocklist` issue is a known problem with the beta Pi-Hole image for `armhf` and is probably why they haven't released any official support for the pi-zero yet.

https://github.com/pi-hole/docker-pi-hole/pull/506

I expect this is also why you are experiencing problems after the upgrade, since you are getting `SIGSEGV` when trying to start the Pi-Hole service, so by upgrading you probably received the incorrect architecture binary.

I don't have a pi-zero myself to test with, but maybe someone in the comments on the issue above has suggestions for getting version 4.4 to run properly?

caiorg, 2021-01-20:
Hey @klutchell!

Thx for the quick reply.

I just did a test with [this image](https://hub.docker.com/r/lewixlabs/pihole-pizero/tags?page=1&ordering=last_updated), changing the line 14 of Dockerfile.rpi and, though it's running PiHole 4.3.1 it is working flawlessly and even noticeably faster.

I'm a newbie with this so I'm on a trial and error basis.

Strangely enough, if I follow that tutorial I've mentioned earlier (don't know if I can post the url here), everything works fine. The only thing I'm left without if I follow it is Balena Cloud.

klutchell, 2021-01-20:
Hey @caiorg, I have notifications enabled on the official docker Pi-Hole repo so if they ever release official images for `armhf` I will pull them into this project as soon as possible!

caiorg, 2021-01-20:
That would be amazing! I'll be glad to help testing.

Since there's a solution needed on their side (official docker pihole), I'll close this and hope they release a fix 😄 .

Thx again for your time @klutchell !

closed 2021-01-20 by caiorg

atwoodjw, 2021-10-01:
@klutchell This may have happened? https://github.com/pi-hole/docker-pi-hole/pull/506#issuecomment-932383444