klutchell/balena-pihole #60: No external screen detected in balenablocks/fbcp causing pihole container to exit

toastyghost710, 2021-01-20:
Starting a new issue as per @klutchell request. The new implementation of fbcp seems to be causing the pihole container to KILL all process and exit because no display is detected by fbcp.

OS Version:
> ID="balena-os"
NAME="balenaOS"
VERSION="2.65.0+rev1"
VERSION_ID="2.65.0+rev1"
PRETTY_NAME="balenaOS 2.65.0+rev1"
MACHINE="raspberrypi3-64"
VARIANT="Development"
VARIANT_ID="dev"
META_BALENA_VERSION="2.65.0"
RESIN_BOARD_REV="7105b35"
META_RESIN_REV="88025c4"
SLUG="raspberrypi3-64"

Here are the logs of what is happening during startup:
> 20.01.21 11:38:55 (-0600) Downloading delta for image 'registry2.balena-cloud.com/v2/cadbb7fd82bca3b0d30fb510e2db7481@sha256:d0556bf4f6fd18ac832635c1d4b3be12fcf04f6fe71554ee5456f8a4b3764b18'
20.01.21 11:39:46 (-0600) Downloaded image 'registry2.balena-cloud.com/v2/cadbb7fd82bca3b0d30fb510e2db7481@sha256:d0556bf4f6fd18ac832635c1d4b3be12fcf04f6fe71554ee5456f8a4b3764b18'
20.01.21 11:39:47 (-0600) Installing service 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'
20.01.21 11:39:47 (-0600) Killing service 'pihole sha256:bc413244e6892e904f1255585a45a179e695184e7a8e27a562a0088b6f66295b'
20.01.21 11:39:47 (-0600) Installed service 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'
20.01.21 11:39:47 (-0600) Starting service 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'
20.01.21 11:39:47 (-0600)  pihole  Stopping lighttpd
20.01.21 11:39:47 (-0600)  pihole  Stopping pihole-FTL
20.01.21 11:39:47 (-0600)  pihole  Stopping cron
20.01.21 11:39:47 (-0600)  pihole  [cont-finish.d] executing container finish scripts...
20.01.21 11:39:47 (-0600)  pihole  [cont-finish.d] done.
20.01.21 11:39:47 (-0600)  pihole  [s6-finish] waiting for services.
20.01.21 11:39:48 (-0600)  pihole  [s6-finish] sending all processes the TERM signal.
20.01.21 11:39:49 (-0600) Started service 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'
20.01.21 11:39:49 (-0600)  fbcp  Display not detected, exiting...
20.01.21 11:39:49 (-0600)  fbcp  Display not detected, exiting...
20.01.21 11:39:51 (-0600) Service exited 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'
20.01.21 11:39:51 (-0600)  pihole  [s6-finish] sending all processes the KILL signal and exiting.
20.01.21 11:39:52 (-0600) Service exited 'pihole sha256:bc413244e6892e904f1255585a45a179e695184e7a8e27a562a0088b6f66295b'
20.01.21 11:39:52 (-0600) Killed service 'pihole sha256:bc413244e6892e904f1255585a45a179e695184e7a8e27a562a0088b6f66295b'
20.01.21 11:39:53 (-0600) Deleting image 'registry2.balena-cloud.com/v2/10aeed7798b9e8488e20e83d28b5ecdb@sha256:a955e2efd8e5fa08e9a221514fa6760d44ea48e50a5284fd70daefc5d2579636'
20.01.21 11:39:53 (-0600) Deleted image 'registry2.balena-cloud.com/v2/10aeed7798b9e8488e20e83d28b5ecdb@sha256:a955e2efd8e5fa08e9a221514fa6760d44ea48e50a5284fd70daefc5d2579636'

klutchell, 2021-01-20:
@toastyghost710 thanks for the details! For some reason I can't reproduce on my slightly older OS, so would you be willing to help me troubleshoot this?

Does the behaviour change if you remove the following line from `docker-compose.yml`?

https://github.com/klutchell/balena-pihole/blob/ee2d4bff3b58736e7b28a0e067aa35d7655f552b/docker-compose.yml#L26

Worst case if we can't figure it out, we can change the `fbcp` container to sleep rather than exit when a screen isn't detected. But the current behaviour should be working so I'd like to do a bit of digging if you don't mind.

toastyghost710, 2021-01-20:
@klutchell that appears to have done the trick. Pi-Hole was able to update gravity and start up and then once no display was detected the fbcp container exited.

> 20.01.21 12:53:56 (-0600)  pihole    [â] Pi-hole blocking is enabled
20.01.21 12:53:56 (-0600)  pihole    Pi-hole version is v5.2.4 (Latest: v5.2.4)
20.01.21 12:53:56 (-0600)  pihole    AdminLTE version is v5.3.1 (Latest: v5.3.1)
20.01.21 12:53:56 (-0600)  pihole    FTL version is v5.5.1 (Latest: v5.5.1)
20.01.21 12:53:56 (-0600)  pihole  [cont-init.d] 20-start.sh: exited 0.
20.01.21 12:53:56 (-0600)  pihole  [cont-init.d] done.
20.01.21 12:53:56 (-0600)  pihole  [services.d] starting services
20.01.21 12:53:56 (-0600)  pihole  Starting pihole-FTL (no-daemon) as root
20.01.21 12:53:56 (-0600)  pihole  Starting crond
20.01.21 12:53:56 (-0600)  pihole  Starting lighttpd
20.01.21 12:53:56 (-0600)  pihole  Starting PADD
20.01.21 12:53:56 (-0600)  pihole  [services.d] done.
20.01.21 12:53:59 (-0600) Downloaded image 'registry2.balena-cloud.com/v2/2842525dacddf2227d42be4fdfe358fb@sha256:d0556bf4f6fd18ac832635c1d4b3be12fcf04f6fe71554ee5456f8a4b3764b18'
20.01.21 12:53:59 (-0600) Installing service 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'
20.01.21 12:53:59 (-0600) Installed service 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'
20.01.21 12:53:59 (-0600) Starting service 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'
20.01.21 12:54:02 (-0600) Started service 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'
20.01.21 12:54:02 (-0600)  fbcp  Display not detected, exiting...
20.01.21 12:54:02 (-0600)  fbcp  Display not detected, exiting...
20.01.21 12:54:03 (-0600) Service exited 'fbcp sha256:44ce3640dff23af0aa9f0f87c0bce59950fe226f540bf4da51bd01addd04e0b9'

klutchell, 2021-01-20:
@toastyghost710 thanks for the help! I pushed the fix to main.

closed 2021-01-20 by klutchell