klutchell/balena-pihole #61: Why the removal of Raspberry Pi 2?

tsialex, 2021-01-20:
Why the reason for the removal of Raspberry Pi 2? I have balena-PiHole 5.4 running it on two Pi 2 and runs fine, even with the slightly older balena kernel for it.
[image]

klutchell, 2021-01-20:
@tsialex only because I don't have an Raspberry Pi 2 in order to test it!

If you can confirm that it works, which it seems you can, I'd be happy to add it back to `balena.yml` and on the README. Or you can open a PR for the change if you'd like it sooner?

tsialex, 2021-01-20:
I can confirm that balena-pihole on the RPI2 works fine with the current 12.3.0 supervisor, no problems at all with two Pi2. I'll test a little more and then open the PR, thx.

toastyghost710, 2021-01-20:
I can second that @tsialex my backup pihole installation is on a RPI2 and even with the most recent update to v5.5.1 there appears to be no issues.

[image]

closed 2021-01-21 by klutchell