klutchell/balena-pihole #64: Problems after FBCP changes

eiddor, 2021-02-18:
Hi there,

I'm having a couple of issues that I _believe_ started after the recent FBCP changes.  I can't be 100% sure because they problems didn't actually appear until after a power cycle (recent power issues in TX).

I'm having two issues:

1) My LCD no longer displays anything (black screen).  This is the official RPi touch screen with the same configuration that I've had in Balena for a couple of years.

The only related thing in the logs is this.  

```18.02.21 16:35:40 (-0600)  fbcp  WARNING: FBCP_DISPLAY variable not set.
18.02.21 16:31:29 (-0600)  fbcp   Set the value if you are using an SPI-based display.
18.02.21 16:31:29 (-0600)  fbcp  Idling...```

There are so many different guides/posts out there about LCD screens, so I hesitate to change anything that was working until I have some idea of where to go, but I have tried setting ```FBCP_DISPLAY``` to ```adafruit-hx8357d-pitft``` and ````adafruit-ili9341-pitft``` to no avail.

Again, I've got the "Official Raspberry Pi 7-inch display" so I don't even know if this is required.  My other settings more or less match what Chris posted [here](https://www.balena.io/blog/assembling-the-official-raspberry-pi-touchscreen/).

2) This issue depends on what my above settings are.  By default (before I started trying to get the display working), all services start and everything looks ok, but pihole does not respond to DNS queries until I restart the container.  No issues in the log, pihole says it starts correctly and that everything is ok.

This has happened to me before, btw - years ago when display settings were all over the map, I would have to manually restart the pihole container after a device reboot so that it would work.

Please let me know if I can share any other settings or logs.

Thanks!
Roddie

klutchell, 2021-02-19:
Hey @eiddor , prior to the recent FBCP changes, did you define a dtoverlay in the format `RESIN_HOST_CONFIG_dtoverlay` in order for your display to work? Is that value still set in your dashboard?

I would suggest to start testing by removing any dtoverlay host config you have set, and then just try all of the available options for `FBCP_DISPLAY` and see if any of them work with your screen. If not, we have other options to try but I want to go this route first.

Regarding the container restart, if that continues to be a problem I would suggest opening a new issue (maybe after we've sorted out the FBCP stuff).

eiddor, 2021-02-19:
Hey @klutchell - Thanks for the reply.

I had ```dtoverlay``` set to ```rpi_display``` but had already tried removing it during troubleshooting.  It's gone again and now my display shows the green logo with the white background.

At this point I've got the following settings still under Device Configuration and now I'm cycling through ```FBCP_DISPLAY``` to see if I can get it to work.

[image]

Good plan on the other issue - I _assume_ once we get the display working, that the pihole container will be happy, but at least when we get to that point we'll know how to break it again :-)

eiddor, 2021-02-19:
@klutchell I tried every value for ```FBCP_DISPLAY``` listed [here](https://github.com/balenablocks/fbcp) without success.  Every option resulted in the white screen/green logo.  

Worth noting for the other issue that we'll come back to: it seems that some combination of removing ```dtoverlay``` and adding ```FBCP_DISPLAY``` makes the pihole container much happier after a reboot.

klutchell, 2021-02-19:
Cool, thanks for testing that. In the short term, there is a branch of this project that supports the old fbcp method. Note that this requires `RESIN_HOST_CONFIG_dtoverlay` and does not use `FBCP_DISPLAY`.

https://github.com/klutchell/balena-pihole/tree/rpi-fbcp

I'm working on supporting additional connection types in one image going forward, but progress has been slow as I only have so much time. One of my two screens also doesn't play well with the new method. Maybe I'll ping you to help with testing when I get to that point? We can also keep this issue open for tracking.

I should also ask, is this your screen type? https://protocentral.com/product/raspberry-pi-official-lcd-7-touchscreen-display/

It looks like it uses DSI for a connection and not the standard SPI drivers and fbcp-ili9341 does not support that type: https://github.com/juj/fbcp-ili9341#what-other-optionsalternatives-do-i-have-to-fbcp-ili9341

All the more reason for me to finish the work that combines both the old and the new mechanisms.

eiddor, 2021-02-19:
Thanks, @klutchell - The old fbcp method branch works as before.  I'll use this for now.  Just have to update the container versions haha

I can confirm that is the screen I'm using - @chrisys wrote about it [here](https://www.balena.io/blog/assembling-the-official-raspberry-pi-touchscreen/) and I ordered it a few days later thinking _"Oh, this will always work with the projects!!"_ hahaha

You can _absolutely_ reach out to me if you need testing help.  

Story - This storm and the resulting power outages exposed two issues for me.  I had my "primary" pihole running on Balena, and the "secondary" pihole running in a VM on my ESXi server.  With the power fluctuations and eventual UPS failures, both failed hard: the dtoverlay/pihole issue one the primary and ESXi mangled its licenses when it lost power on the secondary.  Bottom-line, I spent what little time I had with power/Internet/hotspot building a dedicated pihole RPi that _should_ never fail.

So, yeah, you can use my balena pihole for anything you need :-) 

I'm up and running now and my display is back, so no rush on anything at all.

Do you want me to open another issue for the pihole container problem?  I can't duplicate it anymore since I'm on the special branch, but from what I've been able to glean: if you have ```dtoverlay``` defined in the latest main build, pihole doesn't work without an extra container restart.

klutchell, 2021-02-19:
Let's not open a new issue for that quite yet. I can only work on one thing at a time anyway, and if you can still reproduce it once I've pushed new versions of FBCP then we can open a new issue :)

klutchell, 2021-02-19:
Hey @eiddor , could you try with this branch?

https://github.com/klutchell/balena-pihole/tree/dtoverlay

You'll want to set `FBCP_DISPLAY=dtoverlay` and `RESIN_HOST_CONFIG_dtoverlay=rpi-display`. Your other fields you can probably leave the same.

eiddor, 2021-02-20:
@klutchell Oddly enough, the branch worked for me after the push process, however I did a full shutdown/restart of the device and the screen is now black and pihole only started working after I restarted the container.

```FBCP_DISPLAY=dtoverlay```

Defined DT overlays = ```rpi-display```

klutchell, 2021-02-20:
Maybe a silly question, but you've set `FBCP_DISPLAY=dtoverlay` on the `fbcp` service or globally right?

eiddor, 2021-02-20:
Not a silly question.

I have it set as a device environment variable (in the same place my pihole settings live).  I assumed that is the right place since it got rid of the ```WARNING: FBCP_DISPLAY variable not set.``` way back when :-)

klutchell, 2021-02-20:
Yes, that is one of two correct places. I have another idea, but my wife may kill me if I work on it this weekend. I'll keep you posted ;)

eiddor, 2021-02-20:
@klutchell Ha!!  No worries, man - Just ping me when you have some time and I'll make myself available.  I've got a ton of work to catch up on from the outages.

klutchell, 2021-02-20:
I pushed an old workaround that I thought we didn't need anymore to the `dtoverlay` branch. It fixed both of my screens, can you give it a try when you have a chance?

Stay safe out there!

eiddor, 2021-02-20:
@klutchell Ahh - I remember this workaround well :-) 

1) screen works and survives a full shutdown/restart

2) my pihole container issue appears every time (this is why I remember the workaround)

3) dbus shows exited - I can't remember if this is normal

[image]

klutchell, 2021-02-20:
Sweet! Yes, dbus is supposed to show `Exited`. It runs once to do it's job, then only runs again on restarts, releases, etc.

Good to know the workaround is still required. I'll merge this fix once we've finalized the upstream fbcp PR. I'll keep this ticket open until then.

Can you open a new issue for pihole not working on container start, and include logs from working/not working?

eiddor, 2021-02-20:
Excellent!  

I'll open the new issue here in a minute - There's nothing in the logs to let me know things aren't going to work, but I'll let you double-check me.

closed 2021-03-01 by klutchell