klutchell/balena-pihole #70: Balena PiHole/Unbound and Plex

MadScientist67, 2021-03-01:
Since updating to the build with Unbound, I've lost access to my Plex Server on several devices(Vizio Smart TV, Roku Streaming Stick, PS4, and Xbox360). The rest of the devices on my network (iPads, PCs, Android and iPhones) can still see, connect to the server, and stream shows just fine. I can access the Server using the Roku Media Player on the Streaming Stick, just not the Plex App. 

I found this on a search, but can't figure out how to add the custom domain to Unbound on this build: 

> Hello Pihole Github Community,
> I have a small request for the Pihole Doc's, because I lost a lot of hours with a configuration error and I would like to pass on the resulting solution.
> Pihole with unbound causes a big problem with the Config described in the doc's (https://docs.pi-hole.net/guides/unbound/) when using a plex server in the network (no secure connection can be established via the website "app.plex.tv").
> 
> My request now would be to include this known error in the docs.
> For people who have a plex server in the network, you have to add this entry to the config:
> 
> private domain: plex.direct
> Afterwards the problem is also permanently solved.
> 
> Many thanks for the development of Pihole,
> Xedon

From url: [https://github.com/pi-hole/pi-hole/issues/3461](url)

Apologies if this is not the right way to ask for help. I'm very new to this.

klutchell, 2021-03-01:
Hey @MadScientist67 , thanks for bringing this to my attention.

I don't have this issue on my network, but I can advise how to add a custom domain to unbound. Depending how you have pushed this project to balena your use case may vary.

If you have cloned this repository, or forked and cloned, or just downloaded the source files as a zip, you can change [unbound/unbound.conf](./unbound/unbound.conf) and add the following line near the end under the `private-address:` lines:

From https://github.com/pi-hole/pi-hole/issues/3461

```
    private-domain: plex.direct
```

Then push the changes to your app with the `balena push` command if you are using the balena CLI.

eiddor, 2021-03-01:
@MadScientist67 Also see [this article](https://support.plex.tv/articles/206225077-how-to-use-secure-server-connections/) for more background on why `plex.direct` is different than other mDNS-ish solutions.

MadScientist67, 2021-03-01:
Thanks y'all, 

If I just push using the "Deploy with Balena" button, is there any way to make the change?

klutchell, 2021-03-01:
I'm afraid not, the config changes would be lost on restart since we aren't backing up the unbound.conf.

If you don't want to use the command line tools you could fork this repo, make the change in the Github UI, then Deploy With Balena again but from your repo instead of mine.

klutchell, 2021-03-01:
You also don't have to use Unbound for upstream DNS. You could change your `PIHOLE_DNS_` env var in the balena dashboard to something else, like Cloudflare for example:

`PIHOLE_DNS_=1.1.1.1;1.0.0.1`

eiddor, 2021-03-01:
@klutchell Is it worth adding some common private-domains to `main` or do you think we'll have scale issues?

klutchell, 2021-03-01:
> Is it worth adding some common private-domains to main or do you think we'll have scale issues?

@eiddor I think the definition of "common" private domains might come under some scrutiny but I'm not completely against it. Do you have any others in mind?

eiddor, 2021-03-01:
>  I think the definition of "common" private domains might come under some scrutiny

That's what I meant by "scale" haha

> Do you have any others in mind?

Not off the top of my head, but Plex is one that I would use if I were using Unbound.  I bet it's a pretty common app for the type of user that would run Pi-hole.  Just a though, though - I'm happy to add/PR it.

klutchell, 2021-03-01:
@eiddor Sure, go ahead an PR it. I can name at least one user that will benefit from that.

This opens a wider conversation of whether we should be using Unbound by default, especially if it's difficult to configure. I personally use it but I'm reminded every day that I'm not the typical use case. Do most users care?

What if we pointed to CF by default and left instructions in the README for pointing to Unbound?

eiddor, 2021-03-01:
> Sure, go ahead an PR it. I can name at least one user that will benefit from that.

Will do - @MadScientist67 Give us a few hours and it will be in the main branch.

> This opens a wider conversation of whether we should be using Unbound by default, especially if it's difficult to configure. I personally use it but I'm reminded every day that I'm not the typical use case. Do most users care?

So... I don't think most users will care _unless_ something in Unbound breaks or there's some kind of gap.  Put another way: we're putting a lot of faith into a project we don't manage to no real benefit.

> What if we pointed to CF by default and left instructions in the README for pointing to Unbound?

Personally I think this makes the project more stable because of what I said above.

klutchell, 2021-03-01:
> What if we pointed to CF by default and left instructions in the README for pointing to Unbound?

Okay, I think the README will need a dedicated section for Unbound (re: what/why and how) when we change the defaults to `1.1.1.1;1.0.0.1` in [balena.yml](./balena.yml) and [pihole/Dockerfile](./pihole/Dockerfile).

Do you want to take this one as well or shall I?

eiddor, 2021-03-01:
> Do you want to take this one as well or shall I?

I just learned what Unbound was like 3 1/2 minutes ago, so you should probably take it.  haha

I'll go ahead and add the `private-domain` just to keep the streams separate.

MadScientist67, 2021-03-01:
Thanks guys! I really appreciate your work. Once it's up, I'll push it and let you know how it goes.

klutchell, 2021-03-01:
> to no real benefit

@eiddor I would argue security, privacy, etc but if it's causing friction for new users it should be the advanced option, not the default

eiddor, 2021-03-01:
> I would argue security, privacy, etc but if it's causing friction for new users it should be the advanced option, not the default

You're right - I still think defaulting to CF and then a good write-up about Unbound is probably the safest approach.

PR submitted for Plex.

eiddor, 2021-03-01:
@MadScientist67 You should be able to test now.

MadScientist67, 2021-03-01:
Tested and working on all devices! Thanks y'all.

closed 2021-03-01 by MadScientist67

klutchell, 2021-03-01:
:clap: @eiddor :clap: