klutchell/balena-pihole #73: Balena OS two Questions about Unbound and Dbus

clipse2004, 2021-03-17:
Hi, 

first of all super work! This is the first time that I have no problems with PiHole :) Great work!
However, I have now already read through all the pages but just do not get smarter. Have also tried balena cli but I just do not get it. 
How or where do I find the unbound config? I would like to make a few adjustments, but I just can not get to the config file. Do I have to take the device offline and edit the SD card manually? When I am logged into the Balena CLi via root, I can't find unbound anywhere.
Also from Pihole the setupVars.conf same game.
If I go through the Balena Dashboard on Pihole and open a terminal, then I find the files. However, I can not edit them. 

The other thing is the Dbus service, is that normal that the starts but then stands on status "exited"? 

Am really starting to despair, thank you for your help!

eiddor, 2021-03-17:
Hi @clipse2004 

> How or where do I find the unbound config?

The unbound config can be found in the `unbound/` directory inside of the project - See the files [here](https://github.com/klutchell/balena-pihole/tree/main/unbound)

You'd want to edit `unbound.conf` in that directory and then push Pi-hole to your device with Balena CLI in the traditional way.

Let me know if that makes sense or if you need more detail. There are other ways to do it, but editing that file before you push will make sure that you're using the latest project file and it will persist between reloads, etc. 

> The other thing is the Dbus service, is that normal that the starts but then stands on status "exited"?

That is normal - No worries there.

klutchell, 2021-03-17:
@eiddor , could we treat this as an opportunity to improve the docs?

- how to configure unbound?
- unbound has no shell so web terminal and `balena ssh <uuid> unbound` will not work (for security and size reasons, reduced attack surface, etc)
- dbus should run once on boot and exit

clipse2004, 2021-03-17:
Hello, 
First of all, thank you for your feedback.
I have made the One Click there was nothing with push :) but I should get there.
But again for understanding, if I want to edit the Unbound.conf.
I take the device offline and then edit the config on the SD card via the PC manually and then push that connect? 
Or do I have to reload the project to do this, edit and select "existing device" when creating?
Sorry if I ask something stupid, I'm still not so familiar with the Balena OS :) With Raspbian it goes a little differently.
Just a tip, I can imagine that such a question comes more often, grad with new users. I would prevent the and create a very short How Too "for dummies" 😁 to it.

big thanks!

klutchell, 2021-03-18:
Hey @clipse2004 , you won't be able to change unbound.conf by editing the SD card. There are a couple ways to edit that file while the app is running, but neither of them will persist through a restart so I won't bother sharing those hacks here.

The proper way to edit your unbound.conf is as follows:
1. either clone or download this source repo
2. make the necessary changes to unbound.conf in your local copy
3. [install the balena CLI](https://github.com/balena-io/balena-cli/blob/master/INSTALL.md) and and login with `balena login`
4. [push](https://www.balena.io/docs/reference/balena-cli/#push-applicationordevice) your local changes to your existing app & device with `balena push myOrg/myApp`

We haven't documented how to change that file as we aren't aware of what kind of changes would be needed. The default unbound.conf works for most cases. What exactly are you adding to your unbound.conf?

tatablack, 2021-03-18:
Hi 🙂 Sorry for jumping in, but as I am also changing `unbound.conf`, I thought I could share my requirements.

- Enabling IPV6
  - This required changing `interface`, `do-ip6`, `access-control`
- Configuring a `private-domain`

clipse2004, 2021-03-18:
i want to add these changes:
do-ip6: yes
# increase cache size to utilize more RAM
   msg-cache-size: 128m
   rrset-cache-size: 256m

and if its possible:
remote-control:
        control-enable: yes
        control-interface: XXXXXX
        control-port: 953

and is it possible to edit pihole setupvars.conf, pihole-FTL.conf and 01-pihole.conf? 
i want to add some lines specially cache size =0. I already downloaded the image and found the unbound.conf, but in the pihole folder only the docker and service files :(
I tried pushing earlier today, but the result was:
"Application is ambiguous: balena-pihole".

i understood the instruction that i should download the image, with the cli into the directory where downloaded and then
"balena push balena-pihole"
that was actually so far correct or not? 

I would like to apologize again if I arrive here with the extreme noob question :)

klutchell, 2021-03-18:
> I tried pushing earlier today, but the result was: "Application is ambiguous: balena-pihole".

You'll need to provide your organization name, which is most likely just your username if you click your user dropdown in the top right of the dashboard. For me it is `balena push gh_klutchell/balena-pihole`.

> and is it possible to edit pihole setupvars.conf, pihole-FTL.conf and 01-pihole.conf?

Setup vars should be provided via the balena dashboard as per https://github.com/klutchell/balena-pihole#device-variables or the official [Pi-hole Docker repo](https://hub.docker.com/r/pihole/pihole//) since we are using that image. I don't recommend changing any of those files manually as the Pi-hole startup scripts will overwrite them.

eiddor, 2021-03-18:
> @eiddor , could we treat this as an opportunity to improve the docs?
> 
> * how to configure unbound?
> * unbound has no shell so web terminal and `balena ssh <uuid> unbound` will not work (for security and size reasons, reduced attack surface, etc)
> * dbus should run once on boot and exit

Absolutely - I'll start on some of it this weekend, but I may need you to give me some Unbound input since I've never actually used it.  I can cite the `private-domain: plex.direct` setting that we added along with others in this thread.

eiddor, 2021-03-18:
> You'll need to provide your organization name, which is most likely just your username if you click your user dropdown in the top right of the dashboard. For me it is balena push gh_klutchell/balena-pihole.

Is this strictly true?  For me, "Application Name" refers the name of my application in the Balena console.  In my case below it's "pihole"

[image]

klutchell, 2021-03-18:
@eiddor  Recently we introduced the orgs feature, and since `balena-pihole` is also a public app owned by the balenaHub org, you'll need to specify your "org" or username in front of the app name to avoid `Application is ambiguous: balena-pihole`.

```
$ balena push balena-pihole
Application is ambiguous: balena-pihole

$ balena push gh_klutchell/balena-pihole
[Info]       Starting build for balena-pihole, user gh_klutchell
...
```
If your app name is not duplicated by an official public app, then this isn't required but still recommended.

eiddor, 2021-03-18:
> @eiddor Recently we introduced the orgs feature, and since balena-pihole is also a public app owned by the balenaHub org, you'll need to specify your "org" or username in front of the app name to avoid Application is ambiguous: balena-pihole.

Oh wow -- good to know!  Thank you! 

I guess I've been lucky so far with my applications named `pihole` `pihole-broken` `pihole-almost-fixed` `pihole-oh-crap`. :-) 

I always just assumed it tied the account to the application via the `login` process.

klutchell, 2021-03-18:
https://github.com/balena-io/balena-cli/issues/1893

klutchell, 2021-03-18:
> but I may need you to give me some Unbound input

We definitely shouldn't go into detail of how to customize Unbound configuration, just a quick blurb on how to edit files in that service since we should be directing them to the official Unbound docs for config questions.

So I think we are looking for "How to make edits to unbound.conf and a-records.conf".

eiddor, 2021-03-18:
> So I think we are looking for "How to make edits to unbound.conf and a-records.conf".

Perfect - I think I can handle that and I'll get going on it soon.  I'll remember to add the new push org caveat as well.

(Also, we may want to get [this post](https://www.balena.io/blog/deploy-network-wide-ad-blocking-with-pi-hole-and-a-raspberry-pi/) updated with the push change.  It's the one I always come back to when I forget how to use balena CLI)

I guess it goes without saying that we can't customize things with "One Click?"  I've never used it.

klutchell, 2021-03-18:
> I guess it goes without saying that we can't customize things with "One Click?" I've never used it.

The Deploy with Balena (DwB) button is awesome and allows setting env vars when creating a new app, but sends the source code straight from Github to the balena builders so there's no source code checkout to make advanced changes. However after deployment there's nothing stopping you from sending a new release to the same app.

eiddor, 2021-03-18:
> The Deploy with Balena (DwB) button is awesome and allows setting env vars when creating a new app, but sends the source code straight from Github to the balena builders so there's no source code [...]

I'm going to have to give that a try next time something breaks and I have to spin-up a test application!  haha

clipse2004, 2021-03-18:
hello all, 
holy shit, it's really totally easy to push a new release! However, strangely enough, I only have problems with the unbound.conf. 
I downloaded the zip file balena-pihole-master and set ipv6 to yes in the unbound.conf.
the whole thing then times test pushed -> does not run.

Can it be that there is a difference in the One Button deploy and the zip balena-pihole-master? Because I thought that this One Button version is only pushed automatically.

eiddor, 2021-03-18:
> the whole thing then times test pushed -> does not run.

Can you explain what you mean by this?

clipse2004, 2021-03-18:
on the first try, unbound could not connect at all. The second time, the pushing stopped when downloading "fbcp" or threw an error after a few minutes.
At the last try finally came the message "can not read unbound.conf".
I have attached the unbound config for fun, maybe you will notice an error. 
Changed is now only the interface, ipv6 yes, increase ram and at the bottom forwarded zone. 
otherwise i didn't change anything on the github release, the rest is all done via the fleet.
currently I'm back to the first release, there everything runs so far (without changes). Can you actually delete pushed release somehow in the dashboard? 

btw you would not have fun in addition or alternatively to add a container with cloudflared (DoH) or? :D

big thx
[unbound.txt](https://github.com/klutchell/balena-pihole/files/6166930/unbound.txt)
(don't be surprised, i had to change the file to a txt to be able to upload it)

klutchell, 2021-03-18:
The file you attached looks fine to me. Did you see that error in the dashboard or in your balena push logs? Can you include the logs where you see the error about the config?

If you want to add a cloudflared container I suggesting forking this repo and replacing unbound with cloudflared, rather than providing both images when only one is used at a time.

clipse2004, 2021-03-18:
> The file you attached looks fine to me. Did you see that error in the dashboard or in your balena push logs? Can you include the logs where you see the error about the config?
>
> If you want to add a [...]

Dont have any logs :/ but later I want to switch to an other pi and load unbound conf again. I will save log this time :)
@eiddor did you change IPv6 back to no? This was exact one of my errors too!

eiddor, 2021-03-18:
> @eiddor did you change IPv6 back to no? This was exact one of my errors too!

I just pushed with these settings:

```
# increase cache size to utilize more RAM
msg-cache-size: 128m
rrset-cache-size: 256m

interface: ::0

do-ip6: no
```

No errors, but of course Unbound isn't going to respond to IPv6 queries with that setting.

What I am not sure of is how Balena works with IPv6 in containers.

clipse2004, 2021-03-19:
So now it worked without setting IPv6.
But there is a new unbound Version released 13.1 this should fix this problem.
Is it possible to update unbound? Can I do it by myself?

eiddor, 2021-03-19:
> But there is a new unbound Version released 13.1 this should fix this problem.
> Is it possible to update unbound? Can I do it by myself?

You _can_ but we pull Unbound from @klutchell's repo, so I'd wait until he updates it in this project.

If you really want to try it, edit `Dockerfile` in the `unbound/` directory and change:

`FROM klutchell/unbound-dnscrypt:1.13.0`

to 

`FROM klutchell/unbound-dnscrypt:1.13.1`

And then push.  I'm not sure if there's a reason he hasn't updated yet, though.

klutchell, 2021-03-19:
> And then push. I'm not sure if there's a reason he hasn't updated yet, though.

No reason, just haven't gotten around to it yet. Feel free to PR!

eiddor, 2021-03-19:
Will do!

clipse2004, 2021-03-19:
Tried it today at noon, would not have thought that it works so easily :)
I now successfully installed unbound 1.13.1, but the ipv6 problem still exists. 
now i read that it seems to have something to do with the outgoing connections. 
I will adjust this later in config and test it:

	# specify the interfaces to send outgoing queries to authoritative
	# server from by ip-address. If none, the default (all) interface
	# is used. Specify every interface on an 'outgoing-interface:' line.
	# outgoing-interface: 192.0.2.153
	# outgoing-interface: 2001:DB8::5
	# outgoing-interface: 2001:DB8::6

of course this will be revised :)

Tell me, how can I delete pushed releases in the balena dashboard? 
Slowly become the really many :D

klutchell, 2021-03-19:
@clipse2004 There shouldn't be any need to remove releases in your dashboard. Note that the releases do not live on the device, only in balenaCloud for historical reasons.

eiddor, 2021-03-19:
I didn't even know you could see past releases :-) 

Always learning.

> I now successfully installed unbound 1.13.1, but the ipv6 problem still exists.

@clipse2004 FYI, I've submitted a PR to upgrade the main project to Unbound 1.13.1 - Once it gets reviewed/approved, you can sync with it.

eiddor, 2021-03-21:
> So I think we are looking for "How to make edits to unbound.conf and a-records.conf"

Just sat down to work on this and realized that we already had this in the doc haha.  Doing the other `README.md` updates now.

clipse2004, 2021-03-22:
@eiddor did you try "ipv6 -> yes" with unbound? for me its not working
the unbound problem with the ports is i think more a problem with balena os. the script works great. 
I think we can close the issue then or what do you think ?

eiddor, 2021-03-22:
@clipse2004 I did try it and got the results that I posted [here](https://github.com/klutchell/balena-pihole/issues/73#issuecomment-802287856).  

I'm not really up to speed on how balena/Docker handle IPv6 in a container, though, but I agree that's probably where the issue lies.

klutchell, 2021-03-22:
@clipse2004 Can you please provide logs of the errors you receive when enabling IPv6?

clipse2004, 2021-03-23:
@klutchell we already postet above, the error message is: 
18.03.21 15:28:58 (-0500)  unbound  [1616099338] unbound[1:0] info: start of service (unbound 1.13.0).
18.03.21 15:28:58 (-0500)  unbound  [1616099338] unbound[1:0] error: udp connect failed: Address not available for 2001:500:2::c port 53
18.03.21 15:28:58 (-0500)  unbound  [1616099338] unbound[1:0] error: udp connect failed: Address not available for 2001:500:a8::e port 53
18.03.21 15:28:58 (-0500)  unbound  [1616099338] unbound[1:0] error: udp connect failed: Address not available for 2001:503:ba3e::2:30 port 53
18.03.21 15:33:22 (-0500)  unbound  [1616099602] unbound[1:0] error: udp connect failed: Address not available for 2001:500:12::d0d port 53
but with this settings pihole doesnt resolv any dns querys.
i also don't understand why he wants to bind port 53 when we have set 5053.
I just tried for fun another version that I found: 
https://github.com/aperim/balena-pihole-unbound
here the ipv6 setting works without problems, but i can't find the difference in the settings :( but i know too little about this. 
but i would rather use your version, maybe you can see a difference especially in the unbound config. 
If you need more data, let me know then I let the other pi again with the ipv6 config run. At the moment it is running without setting ipv6.

eiddor, 2021-03-23:
@clipse2004 I had a quick glance at that project and my _guess_ is that it's working because they're running the Unbound container in host mode.

If you want to test with this project, add `network_mode: host` to the `docker-compose.yml' file for the Unbound container, and comment out the ports.  \

Like this:

```
  unbound:
    build: ./unbound
    privileged: true
#    ports:
#      - "5053:5053/tcp"
#      - "5053:5053/udp"
    network_mode: host
```

Then push, etc.

I can't guarantee that this won't cause other issues (@klutchell will probably have flashbacks from a few weeks ago), but it would make sense that this would "fix" it even if it wasn't the actual solution.

klutchell, 2021-03-23:
@clipse2004 Your logs are clear that Unbound is trying to bind to 53 on IPv6, but that port is in use by balena OS so we need to see why your unbound.conf isn't working as expected.

Can you try removing `port: 50531` and instead specify the port in the interface lines?

```
        interface: 0.0.0.0@5053
	interface: ::0@5053
````

eiddor, 2021-03-23:
I don't think this is a binding issue - I've done some Googling and it appears to be more of an upstream issue ("udp connect failed" should be our first clue) and the messages are purely cosmetic.  See [here](https://github.com/MatthewVance/unbound-docker/issues/55).

It also appears to have been fixed in Unbound 0.13.1 per this post: https://github.com/NLnetLabs/unbound/issues/358#issuecomment-736300186

Running it here and I only see the message twice:

```
22.03.21 20:37:55 (-0500)  unbound  [1616463475] unbound[1:0] error: udp connect failed: Address not available for 2001:500:2::c port 53
22.03.21 20:37:55 (-0500)  unbound  [1616463475] unbound[1:0] error: udp connect failed: Address not available for 2001:500:1::53 port 53
```
The problem is that I have no reliable way to test IPv6 resolution.

clipse2004, 2021-03-23:
so they can't be cosmetic. If I look in Pihole under Tail Pihole.log, then you can see there that he does not answer the ipv6 querys. 
Normally a reply message comes back, but in this case nothing, instead the "error message" comes in the log. (this message comes with every ipv6 query)

Does no one use ipv6 from you?

clipse2004, 2021-03-23:
you were right, if i set unbound to 
network_mode: host
then I have no error messages at all and it responds to the querys.

are there any security implications or similar when running in host mode?

klutchell, 2021-03-23:
I personally don't use IPv6 on my LAN, and no one else has reported issues with it so I don't know how many others may be trying to use it.

Looking at your config again it appears you don't have any IPv6 addresses listed in `access-control:`.

Can you also confirm that you're using the latest 1.13.1 in unbound/Dockerfile?

https://github.com/klutchell/balena-pihole/blob/7fb138a31766764f9177fc7c86480a563a991f1d/unbound/Dockerfile#L1

clipse2004, 2021-03-23:
yep is the latest, have it again pushed directly after you have released the.
No in the access control I have nothing entered, I can but like to try. 
What should be in there though?
So config I have found the following:

	# access-control: 0.0.0.0/0 refuse
	# access-control: 127.0.0.0/8 allow
	# access-control: ::0/0 refuse
	# access-control: ::1 allow
	# access-control: ::ffff:127.0.0.1 allow

eiddor, 2021-03-23:
> are there any security implications or similar when running in host mode?

It pretty much opens up all ports and IPs on the host to the container, which can leave you open to vulnerabilities in the container - It's generally not a good practice, but can also be required to workaround port conflicts (pihole in this project runs in host mode for this reason).

If you want to watch @klutchell and I lose hair while discussing it, see [here](https://github.com/klutchell/balena-pihole/issues/66).

Ok, so I just figured something out - When in bridge mode (the opposite of host mode), the Pi-hole container doesn't get an IPv6 address, but when in host mode, it inherits the address from the host.  

I assume this is true of the Unbound container, too, but can't prove it without shell access.  If Unbound doesn't have an IPv6 address, then it won't work with IPv6.

So, we're back to a balena/Docker question: how do we get an IPv6 address to a container while in bridge mode.

I'll do some digging.

klutchell, 2021-03-23:
The requirement for Docker is here: https://docs.docker.com/config/daemon/ipv6/

I assume we don't have these enabled in balena machine by default, but I can ask internally and see if we have it documented already.

klutchell, 2021-03-23:
Also, can you clarify why you require IPv6 between Pi-hole and Unbound? As long as Pi-hole is able to resolve IPv6 requests from your network, which it should be able to do in host networking mode, why does it matter if Pi-hole --> Unbound is IPv4?

Unless you plan on pointing devices to Unbound directly but since it's on a non-standard port that will be difficult to support.

klutchell, 2021-03-23:
I did some experimentation to enable an IPv6 bridge but it seems to be stuck on creating the network without a clear error. Likely because my ipam config is incorrect. The testing branch/commit is here.

https://github.com/klutchell/balena-pihole/commit/c020cb4a3c9286b4952d5a2475a6652517d07293

The Docker documentation is clear that a subnet must be provided either in the daemon args or in the network definition to avoid this issue (I saw this when I left out the ipam config).

https://github.com/moby/moby/issues/29443

I don't have an IPv6 network to test with, but I also don't see the need for inter-service communication. I think your best bet is to run Unbound in host network mode and avoid depending on Docker features that require advanced configuration.

clipse2004, 2021-03-27:
Thanks guys for your help :)

closed 2021-03-27 by clipse2004