klutchell/balena-pihole #82: Curl error - openFleets?

deornoth, 2021-05-28:
Pihole service is failing to start due to the recent changes to the 10-custom.sh.  

```
# openFleets: configure hostname
curl -X PATCH --header "Content-Type:application/json" \
   --data "{\"network\": {\"hostname\": \"$BALENA_HOSTNAME\"}}" \
   "$BALENA_SUPERVISOR_ADDRESS/v1/device/host-config?apikey=$BALENA_SUPERVISOR_API_KEY"
```
I don't know what openFleets is. Even after finding and reading the blog post on it I don't understand. 

28.05.21 09:14:28 (-0500)  pihole  [cont-init.d] executing container initialization scripts...
28.05.21 09:14:28 (-0500)  pihole  [cont-init.d] 10-custom.sh: executing... 
28.05.21 09:14:28 (-0500)  pihole  curl: (3) URL using bad/illegal format or missing URL
28.05.21 09:14:28 (-0500)  pihole  [cont-init.d] 10-custom.sh: exited 3.
28.05.21 09:14:28 (-0500)  pihole  [cont-finish.d] executing container finish scripts...
28.05.21 09:14:28 (-0500)  pihole  [cont-finish.d] done.
28.05.21 09:14:28 (-0500)  pihole  [s6-finish] waiting for services.
28.05.21 09:14:28 (-0500)  pihole  [s6-finish] sending all processes the TERM signal.
28.05.21 09:14:31 (-0500)  pihole  [s6-finish] sending all processes the KILL signal and exiting.

klutchell, 2021-05-28:
Hey there, this CURL command is just to set the hostname of the device to `pihole` for convenience. I might add an optimization to skip this step if `BALENA_HOSTNAME` is empty.

Could you tell me what the value of the following env vars in the container?

```
env | grep -E 'BALENA_SUPERVISOR_ADDRESS|BALENA_HOSTNAME'
```

I could add a condition to not fail everything if that command doesn't work, but I would like to figure out why it's not working on your device first.

deornoth, 2021-05-28:
`env | grep -E 'BALENA_SUPERVISOR_ADDRESS|BALENA_HOSTNAME'`
returns empty.

Where do I set BALENA_HOSTNAME? Everywhere I try to add it says _"Variables beginning with BALENA_ or RESIN_ configure balenaOS features. They must be named as per the documentation and managed on the Configuration page."_ I already have a DEVICE_HOSTNAME variable set in the device variables.

klutchell, 2021-05-28:
That's very odd, both of those env vars should have values. Did you run that in the shell of the `pihole` service, or on the host (they don't exist on the host, but they should be present in the container).

Does your docker-compose file and Dockerfile have these values set?

https://github.com/klutchell/balena-pihole/blob/298da1a78c17e32bd85b0dbb25b4cc36c9b0b165/docker-compose.yml#L19-L20
https://github.com/klutchell/balena-pihole/blob/298da1a78c17e32bd85b0dbb25b4cc36c9b0b165/pihole/Dockerfile#L33

deornoth, 2021-05-28:
Alright, I did not have the io.balena.features.supervisor-api: 1 in the pihole service of my docker-compose.yml file. It was under another service. I added it to the pihole service and tried another build. I'm not getting the curl error now, but the pihole service is just rapidly respawning over and over. I was able to quickly get into the container before it died and grab this.

```
root@pihole:/usr/src/app# env | grep -E 'BALENA_SUPERVISOR_ADDRESS|BALENA_HOSTNAME'
BALENA_HOSTNAME=pihole
BALENA_SUPERVISOR_ADDRESS=http://127.0.0.1:48484
```
Which was not there before. I had to pin to an older release to get the service back up. There are no obvious errors in the logs, so I'm not sure why it's caught in a restart loop.

klutchell, 2021-05-28:
> I did not have the io.balena.features.supervisor-api: 1 in the pihole service of my docker-compose.yml file

That sounds like you have some merge issues in your local working copy. Can you reproduce this with a new clone in a clean workspace? If your files are out of sync or merged incorrectly we could run into any number of issues that I would never see otherwise.

deornoth, 2021-05-28:
Well, io.balena.features.supervisor-api: 1  _IS_ in the balena-pihole/docker-compose.yml file, but I followed the [multi-container tutorial ](https://www.balena.io/blog/two-projects-one-device-turn-your-raspberry-pi-into-a-multitool/) and created my own docker-compose.yml that I'm building from. So I had the supervisor feature under another service instead of pihole.  Not sure if that makes sense, or if I'm explaining it correctly. 

It's been working fine for months, just decided to refresh everything today. I've run out of tinkering time today. I'll try starting fresh tomorrow. Thanks for the quick responses.

klutchell, 2021-05-28:
Okay, keep me posted. I merged the change to avoid failing the service when the hostname cannot be set.

deornoth, 2021-06-10:
Apologies for the delayed response. I don't get a ton of free time to tinker on projects like these. You can close this ticket. It seems to be working now. Though I'm not really sure what fixed it.

I updated my balena_cli tools. Updated the BalenaOS on the device. Updated the balena supervisor on the device. Then I pulled the latest git repos for the device and did another push. 

The pihole service once again started rapidly respawning. Something was killing it as soon as it started. I had a custom 10-restrict.conf in my dnsmasq volume. I deleted that file and everything started working. Then I put the file back and restarted the service and it's still working...   ¯\_(ツ)_/¯

closed 2021-06-11 by klutchell