klutchell/balena-pihole #93: Pi-hole and Homeassistant Port 80 Problem

clipse2004, 2021-09-22:
Hi,
sorry must briefly open an issue, for you certainly a klacks :)
On your config and ports I have changed nothing so runs over unbound with 5053.
I would have to change something in the lighttpd because of the webinterface port assignment (port 80).
I have homeassistant running on another Pi and now have problems with the certificate renewal.
in principle i would like that the webinterface does not listen on the port 80 but on eg. 85. on the normal Pi without balena I have always adjusted this via lighttpd.

can you help me there?

big thanks

klutchell, 2021-09-22:
I'll start off by strongly recommending you do NOT expose your Pi-hole dashboard to the internet. So if you have ports open on your router/firewall pointing to your Pi-hole, I would close them.

That being said, I'm using the official Pi-hole Docker image and it looks like they support an environment variable `WEB_PORT` for setting a custom web interface port for lighttpd.

https://github.com/pi-hole/docker-pi-hole/tree/master#advanced-variables
https://github.com/pi-hole/docker-pi-hole/blob/master/bash_functions.sh#L220-L241

clipse2004, 2021-09-22:
yup i agree with you. 
The error was also not at port 80 as suspected, but the addon has spun.
But thanks for the quick help

closed 2021-09-22 by clipse2004