klutchell/balena-pihole #94: Updating WEBPASSWORD has no effect after initial setup

tmigone, 2021-09-27:
Forgot to set `WEBPASSWORD` before first deploy so I added the variable via dashboard and restarted services but logs only say:

```
pihole  ::: Pre existing WEBPASSWORD found
```

Had to SSH into the running container and run `pihole -a -p "$WEBPASSWORD"` to update it.

@klutchell I'm not familiar with PiHole's startup process, do you think adding the above line [here](https://github.com/klutchell/balena-pihole/blob/main/pihole/cont-init.d/10-custom.sh) would work to avoid this?

eiddor, 2021-09-27:
We define it in `pihole/Dockerfile`:

`ENV WEBPASSWORD balena`

We should probably document that.

tmigone, 2021-09-27:
Oh I knew the password was `balena` but if I wanted to change it the env var wouldn't help. Would be cool if changing the env var updates the password.

tmigone, 2021-09-27:
Exactly! I forgot to set it so the pass was `balena`, then I added the env var expecting the it to be set as the new password but that didn't happen. Only way I found to change the password was running `pihole -a -p "$WEBPASSWORD"`

klutchell, 2021-09-27:
> do you think adding the above line here would work to avoid this?

@tmigone Yeah, that would likely work, but I would like to understand the intended behaviour of the upstream project first.

From this issue it appears to be working as intended, and the password will not change on it's own.
https://github.com/pi-hole/docker-pi-hole/issues/643

However, I wonder if we should actually test deleting `setupVars.conf` on each boot to ensure the latest env vars of all types are used? Is there anything in there that we need to persist vs read from the current env vars?

The `pihole -a -p "$WEBPASSWORD"` command is certainly lower risk if you wanted to open a PR :D