klutchell/balena-pihole #101: Only returning ipv6 address sometimes

Shane01638, 2021-11-06:
I have been struggling with an issue for over a year now.  Sometimes pihole will only return an ipv6 address.  This only seems to happen with *.google.com sites.  For example mail.google.com, meet.google.com, etc.  This becomes apparent when these sites/services stop working.  I am pointing the pihole to 2 local Active Directory/DNS servers.  When I run an nslookup against those 2 servers they will return both an ipv6 and ipv4 address, however when I run nslookup against the pihole it will return only an ipv6 address.  Restarting the pihole resolves this issue for a few months typically.  Is there a way to disable ipv6 or is there a better way to resolve this issue?

eiddor, 2021-11-07:
Can you try the following in the pihole service?

`echo "RESOLVE_IPV6=no" >> /etc/pihole/pihole-FTL.conf `
`service pihole-FTL restart`

This isn't a permanent fix, but I'm curious if it resolves it.

Shane01638, 2021-11-07:
When it happens again I will give that a shot.  Thank you!

Shane01638, 2021-11-09:
> @Shane01638 you could also try setting a Device Variable `IPv6` to `false` in your balenaCloud dashboard.
> 
> IPv6	true	<"true"|"false">	For unraid compatibility, strips out all the IPv6 configuration from DNS/Web services when false.
> https://github.com/pi-hole/docker-pi-hole#optional-variables

Thank you, I added that as a device variable and rebooted.  It is still returning an IPv6 as well as an IPv4 address which is fine with me as long as the IPv4 is present.  I'm not sure if that is the desired outcome from adding this device variable.

eiddor, 2021-11-10:
So, I _think_ that setting is more about the addresses to which the services bind vs. how DNS responds to queries.  You don't need to be running IPv6 to resolve IPv6 addresses; it comes down to the type of record your client requests (IPv6 is AAAA).

Shane01638, 2021-11-10:
Ok, it happened again already.  This is the results:

*Note the device variable IPv6 false is already in place.

Connected to pihole service
`Spawning shell...
root@pihole:/usr/src/app# echo "RESOLVE_IPV6=no" >> /etc/pihole/pihole-FTL.conf
root@pihole:/usr/src/app# service pihole-FTL restart
s6-svc: warning: /var/run/s6/services/pihole-FTL/notification-fd not present - ignoring request for readiness notification
root@pihole:/usr/src/app# SSH session disconnected
SSH reconnecting...
Spawning shell...
root@pihole:/usr/src/app# cat /etc/pihole/pihole-FTL.conf | more
PRIVACYLEVEL=0
RESOLVE_IPV6=no
REPLY_ADDR4=0.0.0.0
root@pihole:/usr/src/app# ^C`

I just restarted the pihole service manually, then made sure the entry was still in pihole-ftl.conf after the reboot.

Here is the problem nslookup:
`> mail.google.com
Server:  pi.hole
Address:  192.168.100.100

Non-authoritative answer:
Name:    googlemail.l.google.com
Address:  2607:f8b0:400a:80b::2005
Aliases:  mail.google.com`

Here is after the pihole service was restarted which is what I usually see:
`> mail.google.com
Server:  pi.hole
Address:  192.168.100.100

Non-authoritative answer:
Name:    googlemail.l.google.com
Addresses:  2607:f8b0:400a:807::2005
          142.250.69.197
Aliases:  mail.google.com`

klutchell, 2021-11-10:
What are you using for upstream DNS? Is it the included Unbound server or the default Cloudflare (1.1.1.1) server?

Shane01638, 2021-11-10:
I am using 2 local Active Directory/DNS servers.  When this happens and I run nslookup against them they return both an IPv6 and IPv4 address.

klutchell, 2021-11-13:
Looking at the FTL code I think it's worth trying what @eiddor suggested above.

Log into the Pi-hole service console and run the following:
```
echo "RESOLVE_IPV6=no" >> /etc/pihole/pihole-FTL.conf
service pihole-FTL restart
```
This should last until the next restart of the container and would help answer some questions.

Shane01638, 2021-11-15:
@klutchell I did add that as posted above but the formatting was garbage.  I'll try to clean it up:

`Connected to pihole service

Spawning shell... 
root@pihole:/usr/src/app# echo "RESOLVE_IPV6=no" >> /etc/pihole/pihole-FTL.conf 
root@pihole:/usr/src/app# service pihole-FTL restart 
s6-svc: warning: /var/run/s6/services/pihole-FTL/notification-fd not present - ignoring request for readiness notification 
root@pihole:/usr/src/app# 

(This is where I restarted the service through balena when the command failed, then below verified the entry was still present)

SSH session disconnected SSH reconnecting... 
Spawning shell... 
root@pihole:/usr/src/app# cat /etc/pihole/pihole-FTL.conf | more 
PRIVACYLEVEL=0 
RESOLVE_IPV6=no 
REPLY_ADDR4=0.0.0.0 
root@pihole:/usr/src/app#`

A new error I'm getting now is my unraid server is being rate-limited, likely due to all the docker containers running on it.

klutchell, 2021-11-15:
You can adjust or disable your rate limit with the following FTL var, in the same pihole-FTL.conf.
https://docs.pi-hole.net/ftldns/configfile/#rate_limit

I have a PR upstream to expose these rate limit as and env var going forward.
https://github.com/pi-hole/docker-pi-hole/pull/942

I might do the same for `RESOLVE_IPV6` if this setting works for you.

eiddor, 2021-11-15:
> This is where I restarted the service through balena when the command failed,

Just FYI, the restart command did not fail - That message is just a warning.

closed 2022-01-17 by Shane01638