klutchell/balena-pihole #114: no upstream servers configured

Shane01638, 2022-01-17:
Good day,

Yesterday my pihole stopped resolving DNS queries with a Status of 	OK (cache) (not ready)	Reply REFUSED (0.1ms) for all clients.  Restarting didn't resolve the issue.  I saw it needed an update so I updated with the same result but now pihole had an error DNSMASQ_WARN	Warning in dnsmasq core:
no upstream servers configured
I use local windows DNS servers if that makes a difference.  Using nslookup both of those resolve DNS queries, but the pihole does not.  I have played with different IPs in the DNS1, DNS2, PIHOLE_DNS_ variables.  I'm not sure what has happened.

klutchell, 2022-01-17:
Hey @Shane01638, may I ask how you deployed this project? Are you part of the open fleet (via balena hub) or did you use the deploy button on the readme, or are you pushing new releases with the balena CLI?

Also what is your current value for `PIHOLE_DNS_`? Do you have access to the logs from the container?

Shane01638, 2022-01-17:
I use the balena cloud and updated with the link in the readme.  

The current values of PIHOLE_DNS_ are 192.168.100.15;192.168.100.21 which are my local windows DNS servers.  I do have access to logs I believe.

Shane01638, 2022-01-17:
On the main dashboard under Upstream servers it only lists blocklist and cache no matter what DNS servers I select under settings.  When I save changes in settings it says the settings are updated using 2 DNS servers.

klutchell, 2022-01-17:
Can you share some of the logs from the pihole service startup in the dashboard?

Do you know of anything on your network that may have changed since yesterday?

Shane01638, 2022-01-17:
Is this what you are looking for?

[zpihole_summer-water-17.01.22_10_20_11_(-0900).txt](https://github.com/klutchell/balena-pihole/files/7884243/zpihole_summer-water-17.01.22_10_20_11_.-0900.txt)

Shane01638, 2022-01-17:
When I couldn't get anywhere with the pihole I went through and updated everything.

klutchell, 2022-01-17:
Does it work if you use different upstream servers, like the cloudflare ones `1.1.1.1;1.0.0.1` ?

I don't see any obvious issues in your logs. You may want to try generating a debug log under Tools and see if you notice anything there, maybe share it on the P-hole forums as well since we use the official Pi-hole docker image.

Shane01638, 2022-01-17:
Thanks for your help.  Might just be time to build a new one from scratch but I'll check with the pihole forums.  This one has been running since May 2019.

Shane01638, 2022-01-19:
I have some new information from the pihole team you might be able to help me with:

From your debug log - you have upstream DNS servers configured but Pi-hole isn't listening on your network. Is this install in Docker?

*** [ DIAGNOSING ]: Networking
[✗] No IPv4 address(es) found on the  interface.
[✗] No IPv6 address(es) found on the  interface.

*** [ DIAGNOSING ]: Setup variables
    IPV4_ADDRESS=0.0.0.0
    PIHOLE_INTERFACE=eth0
    PIHOLE_DNS_1=192.168.100.15
    PIHOLE_DNS_2=192.168.100.21

klutchell, 2022-01-19:
Thanks for the additional info @Shane01638!

Is your device using wired or wireless networking? What value do you have set for `INTERFACE`?

Shane01638, 2022-01-19:
Wired, I use a POE hat; INTERFACE and DNSMASQ_LISTENING are both eth0

klutchell, 2022-01-19:
Hmm... this startup snippet should prevent Pi-hole from starting until your desired interface has an IP, so we are getting past that part at least.
https://github.com/klutchell/balena-pihole/blob/15d9c2c97118ae6aadb69509ea2a8b98357c186f/pihole/cont-init.d/10-custom.sh#L8-L12

Shane01638, 2022-01-19:
Is that something that exists or something you would like me to try?  Is there a place I can set a static IP instead of using a DHCP reservation?

Shane01638, 2022-01-19:
It's obviously getting an IP as I can access the web console, balena cloud sees it, and it uploaded the debug logs here: https://tricorder.pi-hole.net/teiBuEab/

klutchell, 2022-01-19:
That snippet I shared is running today, before the Pi-hole service starts.

You can set a static IP by following the guides here, but I don't think that's the issue:
https://www.balena.io/docs/reference/OS/network/2.x/#setting-a-static-ip

klutchell, 2022-01-19:
Unfortunately I don't have access to the diagnostic logs you linked above. Can you attach them here?

Shane01638, 2022-01-19:
Sure thing, I would like to remove them though once you grabbed them
[pihole_debug_log.txt](https://github.com/klutchell/balena-pihole/files/7900225/pihole_debug_log.txt)
.

klutchell, 2022-01-19:
Got them, thanks!

klutchell, 2022-01-19:
Could you try changing the following env vars?

- if you are manually setting `IPV4_ADDRESS` can you unset it?
- `DNSMASQ_LISTENING` should be set to `single`, or `all`, not `local`
- set `PIHOLE_DNS_` to `1.1.1.1;1.0.0.1` or `127.0.0.1#5053;127.0.0.1#5053` just for testing

Shane01638, 2022-01-19:
I was not using IPV4_ADDRESS, I changed DNSMASQ_LISTENING to single and verified it changed in the console as well as changing PIHOLE_DNS_ to 1.1.1.1;1.0.0.1 and verified it changed in the console.  I still have DNS1 and DNS2 set to 1.1.1.1 and 1.0.0.1.   Same results as before, still not working.

klutchell, 2022-01-19:
Thanks! Do you have any value set for `ServerIP`? If yes can you remove it, if no can you set it to your LAN IP and see if it changes the behaviour? It appears unset from your logs and it shouldn't be required... but let's see what happens.

This definitely seems like the issue but I don't know what would make it start happening out of the blue unless something changed on your network or device.
```
*** [ DIAGNOSING ]: Networking
[✗] No IPv4 address(es) found on the interface.
[✗] No IPv6 address(es) found on the interface.
```

klutchell, 2022-01-19:
If you link to your open forum issue I can comment there as well. To answer their questions:

> Yes you are running in docker, with host networking and bind_interfaces set in dnsmasq.conf.

Shane01638, 2022-01-19:
I just posted here on reddit.  Setting ServerIP didn't change anything.  https://www.reddit.com/r/pihole/comments/s7ido3/reply_refused_no_upstream_servers_configured/

klutchell, 2022-01-19:
Well I'm still looking at your debug log, and the specific errors pointed out by reddit come from this line of the diagnostics script:
https://github.com/pi-hole/pi-hole/blob/2ff10fcd0a5464255d9c004541e6b23d7b42662f/advanced/Scripts/piholeDebug.sh#L639

So I expect that if you run the following commands in your container shell, it will return no results. You can also try on your hostOS.
```
ip -4 addr show dev eth0 | awk -F ' ' '{ for(i=1;i<=NF;i++) if ($i ~ '/^inet/') print $(i+1) }'
ip -6 addr show dev eth0 | awk -F ' ' '{ for(i=1;i<=NF;i++) if ($i ~ '/^inet/') print $(i+1) }'
```

On my device the first command returns my local IPV4. I get the same result running it on my hostOS which is expected.

```
root@pihole:/usr/src/app# ip -4 addr show dev eth0 | awk -F ' ' '{ for(i=1;i<=NF;i++) if ($i ~ '/^inet/') print $(i+1) }'
192.168.1.55/24
```

I'm not certain, but perhaps this is related to the IPv6 issues you saw previously in https://github.com/klutchell/balena-pihole/issues/101.

Thank you for your patience, I think we are starting to narrow things down.

Shane01638, 2022-01-20:
In the pihole terminal AND host OS I get IPs for both, the first command is the IPv4 IP that is my pihole so it is correct.

`root@pihole:/usr/src/app# ip -4 addr show dev eth0 | awk -F ' ' '{ for(i=1;i<=NF;i++) if ($i ~ '/^inet/') print $(i+1) }'
192.168.100.100/24
root@pihole:/usr/src/app# ip -6 addr show dev eth0 | awk -F ' ' '{ for(i=1;i<=NF;i++) if ($i ~ '/^inet/') print $(i+1) }'
fe80::40e8:37e0:5be7:4ec/64
root@pihole:/usr/src/app# `

Also, I think I found the pihole forums and created a thread here.  I'll let you know if I get anything useful. https://discourse.pi-hole.net/t/no-ipv4-address-es-found-on-the-interface/52796

One thing I noticed, after this problem started and I went through and updated everything to include the BalenaOS it seems there is now CPU and Memory statistics at the top of Balena Cloud which I don't remember seeing before, anyway I also see this error Danger!The device is undervolted! which I'm not sure how long that might have been a problem.  I don't know how accurate that is but if you think it could be an issue I could try to remove the POE hat and use a standard raspberry pi power supply.  I can say it is only drawing 3.18W of power.

klutchell, 2022-01-20:
The undervoltage likely isn't related. Hopefully the balenaOS update isn't related either but it might be worth running the Device Health Checks and Device Diagnostics under the DIAGNOSTICS side bar button in your balena Dashboard. If any issues pop up there we can investigate further.

I really don't understand how you can get this error in your Pi-hole diagnostics:
```
*** [ DIAGNOSING ]: Networking
[✗] No IPv4 address(es) found on the  interface.

[✗] No IPv6 address(es) found on the  interface.

[i] Default IPv4 gateway: 192.168.100.1
   * Pinging 192.168.100.1...
[✗] Gateway did not respond. (https://discourse.pi-hole.net/t/why-is-a-default-gateway-important-for-pi-hole/3546)
```
But this command works when run manually in your container:
```
$ ip -4 addr show dev eth0 | awk -F ' ' '{ for(i=1;i<=NF;i++) if ($i ~ '/^inet/') print $(i+1) }'
192.168.100.100/24
$ ip -6 addr show dev eth0 | awk -F ' ' '{ for(i=1;i<=NF;i++) if ($i ~ '/^inet/') print $(i+1) }'
fe80::40e8:37e0:5be7:4ec/64
```
Because your diagnostics should look like this (mine):
```
*** [ DIAGNOSING ]: Networking
[✓] IPv4 address(es) bound to the eth0 interface:
    192.168.1.55/24

[✓] IPv6 address(es) bound to the eth0 interface:
    fe80::35a8:4009:7b5a:617a/64
    
[i] Default IPv4 gateway: 192.168.1.1
   * Pinging 192.168.1.1...
[✓] Gateway responded.
```
So everything else aside, it is VERY weird that running the exact same commands manually works, but fails when executed by Pi-hole.

Shane01638, 2022-01-20:
I have no idea.  I can ping the gateway from the pihole as well.  My initial thought was failing SD card that is failing in a very specific place lol.  I'm not sure if reloading the pihole makes sense with the way the containers work.  Since I updated pihole wouldn't the container have been re-installed?  I updated Balena and pihole after these problem started in an effort to resolve them so my thought was if there was an SD card problem whatever missing or corrupt files would have been repaired so to speak.  I have no reason to think there is an SD card problem other than I couldn't think of anything else.  I do need to figure out how to run pihole in unraid.  I guess I could run it in an ubuntu vm, but unraid won't use a vm or docker ip as a dns server I think.

klutchell, 2022-01-20:
The containers would have been recreated when updating Pi-hole but not the data volumes. You can try the Purge Data action from the dashboard to remove the existing volumes and recreate them. Make sure to export your Pi-hole "Teleporter" data first if you want to restore your block lists, etc.

I would suggest running the balena diagnostics and health checks I mentioned to see if it flags any OS issues like media corruption or other service issues. You can share the healthcheck results here.

You can also flash a new SD card and swap them out to see if anything changes, as a last resort.

Shane01638, 2022-01-20:
[devicediagnostics.txt](https://github.com/klutchell/balena-pihole/files/7906945/devicediagnostics.txt)
[devicehealthchecks.json.txt](https://github.com/klutchell/balena-pihole/files/7906953/devicehealthchecks.json.txt)
I ran the health checks and posted, note changing the extension on devicehealthchecks it didn't like json.
I back up the teleporter data and purged data.
It worked immediately upon starting back up!  I restored my data and it still worked!  I then changed the PIHOLE_DNS_ variable back to my local DNS servers and it is still working.  I'm thinking the purge data fixed it :)

klutchell, 2022-01-20:
I do see from your debug log that your configuration files have fewer values than mine. Not sure if it was stale data or something from the migration, but deleting them and recreating them seems to have worked.
```
-rw-r--r-- 1 pihole pihole 66 Jan 17 19:36 /etc/pihole/pihole-FTL.conf
   PRIVACYLEVEL=0
   RESOLVE_IPV6=no
   RATE_LIMIT=0/0
   REPLY_ADDR4=0.0.0.0

*** [ DIAGNOSING ]: contents of /etc/dnsmasq.d

-rw-r--r-- 1 root root 1.3K Jan 17 19:55 /etc/dnsmasq.d/01-pihole.conf
   addn-hosts=/etc/pihole/local.list
   addn-hosts=/etc/pihole/custom.list
   localise-queries
   no-resolv
   cache-size=10000
   log-queries
   log-facility=/var/log/pihole.log
   log-async
   interface=eth0
 ```
 Glad we got it working!

closed 2022-01-20 by klutchell