klutchell/balena-pihole #145: PADD is only using half the display

klutchell, 2022-05-03:
In recent releases by devices with screens are compressing the PADD output and only using the top half of the display.

Unclear at this point if it is related to FBCP, or the balenaOS release, or something else altogether.

eiddor, 2022-05-03:
Hmmmm - I haven't seen this issue in a long time.  I believe the last time I saw it was when I realized that we had a font/font size option.

klutchell, 2022-05-03:
What type of screen do you have, and what balenaOS version are you running? What is your value for `FBCP_DISPLAY` and other HDMI related device configuration?

eiddor, 2022-05-03:
I've got the official RPi 7" (I think?) screen.  FBCP_DISPLAY is dtoverlay, and here are the rest of my settings.

[image]
[image]
[image]

klutchell, 2022-05-03:
I'm running a newer balenaOS release, I wonder if something changed in the upstream display firmware we need to take into account now.

eiddor, 2022-05-03:
Let me upgrade and I'll let you know.

eiddor, 2022-05-03:
Yep - Upgraded to the latest recommended balenaOS release and I've got the same issue.

klutchell, 2022-05-05:
This appears to work with privileged mode, so we are missing either a capability or a device that is required for `console-setup` to actually take effect on `tty1`.

I am opening a PR to restore the ability to use privileged mode upstream. I also intend to figure out what we are missing for console-setup so we can continue to not run as privileged.

closed 2022-05-06 by klutchell