klutchell/balena-pihole #181: reverse proxy not working on lan

mckaycr, 2022-11-07:
Sorry its me again.  I'm trying to apply this change to my deivce: [Using Pi-hole to route your services internally](https://medium.com/@containeroo/using-pi-hole-to-route-your-services-internally-2ff495093718).  I want to pretty much tell pihole that local requests to my domain should be routed to my reverse proxy server instead of external dns servers.  just FYI, I'm not currently using pihole as a local DNS server, I know I could, I just don't feel like it.

eiddor, 2022-11-07:
@mckaycr What problem are you having? It's one thing to setup pihole with custom DNS records (you don't have to do it at the CLI, you can do it in the GUI on the page in the attached screenshot.)

But at the end of the day, pihole needs to be your DNS server if you want it to actually work.

mckaycr, 2022-11-07:
I have one synology server that manages all my reverse proxy stuff both internal and external to my network.  I'd like to leave that in place as is rather than try and configure DNS on my pihole.

eiddor, 2022-11-07:
Oops - forgot to attach the screenshot earlier.

I'm still trying to understand what you mean - Are you actually having an issue doing this?  

The reverse proxies can stay on the Synology, but if you want your internal network devices to resolve www.mckaycr.com (made that up) to the internal IP of the Synology vs. your external IP, you will need to configure a DNS server (like pihole) to handle that.  And then your clients will need to point to pihole in order to resolve it.

[image]

mckaycr, 2022-11-07:
Right, so I did that, and it does resolve www.mckaycr.com (too expensive for me to buy), but it doesn't resolve subdomains that the reverse proxy at mckaycr.com usually serves up.

eiddor, 2022-11-07:
So you're looking for something like this?

```
➜  ~ nslookup
> server 10.39.31.10
Default server: 10.39.31.10
Address: 10.39.31.10#53
> chris.mckaycr.com
Server:		10.39.31.10
Address:	10.39.31.10#53

Name:	chris.mckaycr.com
Address: 10.10.10.1
> www.mckaycr.com
Server:		10.39.31.10
Address:	10.39.31.10#53

Name:	www.mckaycr.com
Address: 10.10.10.1
> foo.mckaycr.com
Server:		10.39.31.10
Address:	10.39.31.10#53

Name:	foo.mckaycr.com
Address: 10.10.10.1
```
[image]

mckaycr, 2022-11-07:
yes, which my synology can do already if I could tell pihole that *.mckaycr.com is 10.10.10.1.  The link I sent basically does this but I think balena doesn't let me edit the files I need to make this entry

mckaycr, 2022-11-08:
thanks, that worked.  I know the correct answer is stop using synology for what pihole can already do, but I'd rather no change whats already working.

eiddor, 2022-11-08:
I'm glad you got it working!  I'm still a bit confused about where you think the Synology's function (reverse proxies, apps in Docker containers) conflicts with pihole's (DNS).  Are you doing DNS on the Synology?

mckaycr, 2022-11-08:
My synology reverse proxy serves up a ton of servers under my domain, not just docker container.  Synology doesn't serve any other roles other than docker and reverse proxy.  When I'm on my LAN and use a fully qualified domain name to browse to a server, for some reason the name isn't getting resolved whenever DNS queries are forwarded through pihole.  You'd think the request would leave the house, go to the internet and come right back in, but it doesn't.  Adding the wildcard config solves this for me.  I could take the reverse proxy down completely I guess and use PiHole, but like I said, I didn't want to make to many changes all at once.

eiddor, 2022-11-08:
Ahhh ok, I think I get it now - There's no reason to change what you're doing with the Synology.  I use a bunch of reverse proxies here myself (on a dedicated ESXi server, in my case.) Reverse proxies and Pihole DNS compliment each other and are not mutually exclusive.  Where Pihole helps, which is what you did last night, is that it can allow your Internal hosts to use the reverse proxy while bypassing public DNS.

closed 2022-11-10 by eiddor