klutchell/balena-pihole #187: stty: standard input: Inappropriate ioctl for device

klutchell, 2022-12-12:
```
12.12.22 11:12:43 (-0600)  pihole    [i] Storing downloaded domains in new gravity database...Starting PADD...
12.12.22 11:12:44 (-0600)  pihole  stty: 'standard input': Inappropriate ioctl for device
12.12.22 11:12:44 (-0600)  pihole  stty: 'standard input': Inappropriate ioctl for device
12.12.22 11:12:44 (-0600)  pihole  /usr/src/app/padd.sh: 939: [: Illegal number: 
12.12.22 11:12:44 (-0600)  pihole  /usr/src/app/padd.sh: 944: [: Illegal number: 
12.12.22 11:12:44 (-0600)  pihole  /usr/src/app/padd.sh: 949: [: Illegal number: 
12.12.22 11:12:44 (-0600)  pihole  /usr/src/app/padd.sh: 954: [: Illegal number: 
12.12.22 11:12:44 (-0600)  pihole  /usr/src/app/padd.sh: 959: [: Illegal number: 
12.12.22 11:12:44 (-0600)  pihole  /usr/src/app/padd.sh: 964: [: Illegal number: 
12.12.22 11:12:44 (-0600)  pihole  /usr/src/app/padd.sh: 969: [: Illegal number: 
12.12.22 11:12:44 (-0600)  pihole  /usr/src/app/padd.sh: 974: [: Illegal number: 
12.12.22 11:12:44 (-0600)  pihole  [â] Error!
12.12.22 11:12:44 (-0600)  pihole      PADD isn't
12.12.22 11:12:44 (-0600)  pihole      for ants!
12.12.22 11:12:44 (-0600)  pihole  
12.12.22 11:12:44 (-0600)  pihole  Stopping PADD
```

_Originally posted by @eiddor in https://github.com/klutchell/balena-pihole/issues/186#issuecomment-1346904543_

eiddor, 2022-12-12:
So, is it a matter of adding the `if [ -t 0 ] ; then` condition to each of those sections?

eiddor, 2022-12-12:
Or rather to lines 935 & 936?

```
if [ -t 0 ] ; then
      console_height=$(stty size | awk '{ print $1 }')
      console_width=$(stty size | awk '{ print $2 }')
else
      something something I don't know what I'm doing
```

klutchell, 2022-12-12:
I think if they switch back to `tput` it would be fine?
https://github.com/pi-hole/PADD/pull/268/files#diff-42c44861235d0baff7c03aea0fe6c7b7cb95f444c3a398c6d96da1290fc6bc01L1059-L1060

closed 2022-12-13 by klutchell