klutchell/balena-pihole #204: Static IP assignment

n1md4, 2023-02-06:
> On your router or DHCP server assign a static IP to your Pi-hole device, and set your clients DNS to the same IP address.

This would not work for me.  I would set as such on my ISP router, but after disabling DHCP on ISP router it would seemingly drop connectivity to the Pi-Hole.  Upon enabling DHCP on ISP router connectivity would resume.  I worked around this by setting a static IP address within resin.io as detailed [here](https://blog.balena.io/how-to-configure-a-static-ip-address-with-resin-io/).

I would have thought more issues would have been raised if this was a wider problem, but perhaps a foot-note is worth adding to the wiki just in case :)

eiddor, 2023-02-06:
The wording is technically incorrect - What we're actually suggesting is that you create a reservation on your DHCP server for your Pi-hole device so that it assigns a consistent address.  The issue is that not every DHCP server (especially ISP-provided ones) actually allow reservations.  We're not actually telling you to turn off DHCP; that would be bad!

I'll try to clean up the wording sometime soon.

n1md4, 2023-02-06:
Thanks.  Appreciate not everyone wants to change DHCP too, but the UI on the Virgin Media Hub 3 is absolutely awful.  I had DHCP on the Pi-Hole enabled, and would then disable DHCP on the ISP router.   No real issue here, can close.  Only posted because it took me a while to resolve my issue.

closed 2023-02-06 by eiddor