klutchell/balena-pihole #223: PADD not showing on balenaOS 2.113.18

klutchell, 2023-04-06:
It may be related to this commit:
https://github.com/balena-os/meta-balena/commit/72706964b8dd308b6c1772fa22c4e89d1680d205

combined with how we attempt to stop the plymouth service here:
https://github.com/klutchell/balena-pihole/blob/37f97489c9718977e6a629e4ade1fcaeeef1f233/pihole/s6-overlay/s6-rc.d/padd/run#L4-L12

klutchell, 2023-04-06:
Some diffs between the systemd unit between 2.101 and 2.113

```
< ExecMainStatus=0
< ExecStart={ path=/bin/plymouth ; argv[]=/bin/plymouth quit ; ignore_errors=yes ; start_time=[Wed 2023-04-05 23:57:54 UTC] ; stop_time=[Wed 2023-04-05 23:57:54 UTC] ; pid=3082 ; code=exited ; status=0 }
< ExecStartEx={ path=/bin/plymouth ; argv[]=/bin/plymouth quit ; flags=ignore-failure ; start_time=[Wed 2023-04-05 23:57:54 UTC] ; stop_time=[Wed 2023-04-05 23:57:54 UTC] ; pid=3082 ; code=exited ; status=0 }
---
> ExecMainStatus=1
> ExecStart={ path=/bin/plymouth ; argv[]=/bin/plymouth quit --retain-splash ; ignore_errors=yes ; start_time=[Thu 2023-04-06 13:26:30 UTC] ; stop_time=[Thu 2023-04-06 13:26:30 UTC] ; pid=449 ; code=exited ; status=1 }
> ExecStartEx={ path=/bin/plymouth ; argv[]=/bin/plymouth quit --retain-splash ; flags=ignore-failure ; start_time=[Thu 2023-04-06 13:26:30 UTC] ; stop_time=[Thu 2023-04-06 13:26:30 UTC] ; pid=449 ; code=exited ; status=1 }
```
```
< OnSuccesJobMode=fail
---
> OnSuccessJobMode=fail
```

klutchell, 2023-04-06:
Another investigation, balenaOS provides some kernel command line arguments in the plymouth services, but they don't appear in `/proc/cmdline`.

klutchell, 2023-04-06:
It may be related to how we quit the plymouth service here
https://github.com/klutchell/balena-pihole/blob/37f97489c9718977e6a629e4ade1fcaeeef1f233/pihole/s6-overlay/s6-rc.d/padd/run#L4-L12

klutchell, 2023-04-06:
For some reason the following commands on the host can "fix" it
```
plymouthd --debug --tty=/dev/tty1
plymouth show-splash
plymouth --quit
```

klutchell, 2023-05-11:
Was working in `2.108.18+rev1`

klutchell, 2023-09-13:
It seems that in addition to disabling getty on development devices, we also need to remove the `--retain-splash` from the `plymouth-quit.service`

```
root@97e5990:~# cat /lib/systemd/system/plymouth-quit.service  
[Unit]
Description=Terminate Plymouth Boot Screen
After=rc-local.service plymouth-start.service systemd-user-sessions.service

[Service]
ExecStart=-/bin/plymouth quit --retain-splash
Type=oneshot
RemainAfterExit=yes
TimeoutSec=20
```

Tested with a systemd drop-in service.

```bash
mount -o rw,remount /
systemctl edit plymouth-quit.service
```
```
[Service]
ExecStart=                   
ExecStart=-/bin/plymouth quit
```

klutchell, 2023-09-15:
Also see https://github.com/balena-os/meta-balena/issues/1081

klutchell, 2023-10-25:
Should be addressed by https://github.com/balena-os/meta-balena/pull/3288

eiddor, 2023-10-25:
Going to rebuild my RPi touchscreen at some point today so hopefully I can test.

klutchell, 2023-10-26:
I expect you are running into [your original issue](https://github.com/klutchell/balena-pihole/issues/66) since I a while ago [I removed this step](https://github.com/klutchell/balena-pihole/commit/1a8550596a5628aebb2c6bbe9eb17eb2c725bbd0#diff-ceab6eb185d1777a7caed3723684c9fda0a25fba3ba9ae8fa02296bb6a32a2d7L18-L22) hoping it was no longer needed.

I guess we need to add that step back for your "special" DHCP network.

eiddor, 2023-10-26:
Yep - looks like that's it.  I enabled STP Portfast on the interface to force it up without an STP calculation and it works fine.

It's not special, it's networking!! :-D

klutchell, 2023-10-26:
Let's keep this issue on-topic please! You can re-open your old issue if you think we should add the DHCP wait back.

Regarding this PADD issue, it seems your device is not affected because the indended fix is in v4.1.1 of balenaOS.

eiddor, 2023-10-26:
Hahaha - Ok, ok.

Hmmm - PADD didn't display until I removed the `--retain-splash` flag, but I'll test again on the RPi4 tomorrow.

klutchell, 2023-10-26:
> and applied your fix

Ah, you applied the manual fix. I missed this. You are affected, and balenaOS v4.1.1 should remove the need for the manual fix.

klutchell, 2023-10-30:
balenaOS v4.1.1 has resolved the issue on my devices

closed 2023-12-14 by klutchell