klutchell/balena-pihole #232: Access denied for tailscale

chrisdeely, 2023-06-21:
I just tried to deploy the most recent version to my RPi 1 B, and ran into an error when building `tailscale` - is there an appropriate version available?
```
[Info]       Building on 3c65b7a
[Info]       No suitable previous release for caching, skipping cache pull
[Info]       Starting to build pihole
[Info]       Starting to build unbound
[Success]    Successfully built service unbound
[Info]       Starting to build tailscale
[Success]    Successfully built service pihole
[Info]       Uploading images
[Success]    Successfully uploaded images
[Error]      Some services failed to build:
[Error]        Service: tailscale
[Error]          Error: pull access denied for bh.cr/klutchell_blocks/tailscale-rpi/1.42.0, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
[Info]       Built on 3c65b7a
[Error]      Not deploying release.
```

klutchell, 2023-06-21:
I haven't published a tailscale block for that architecture unfortunately. Are you running on Raspberry Pi 0?

I've only tested the block with amd64, aarch64, and armv7hf so far. I can try it with rpi/armv5 if you need it.

chrisdeely, 2023-06-21:
I _had_ been running on a Pi Zero W for over a year (after a lot of initial finagling) until it suddenly stopped working. All of the info I can find says pi hole / docker don't play well together on the Zero, so I dusted off my old  RPi v1 B to try that out instead.

I have a forked version of your repo and I removed tailscale & fcbp from the compose file. The build just succeeded, so I'll write back once I confirm if it deploys properly.

chrisdeely, 2023-06-21:
the build deployed successfully, and pihole appears to be running, but I get a 403 Forbidden when attempting to connect to the pihole admin site. is that access something which tailscale would have arranged?

klutchell, 2023-06-21:
> is that access something which tailscale would have arranged?

No, tailscale is a completely optional service. The approach of making a fork and removing the service sounds correct.

Are there any logs from the pihole service that might indicate an issue? If you run the healthchecks in the balena dashboard it might give an indication if something else is wrong on the OS.

If you share your forked repo here I can double check your compose file.

chrisdeely, 2023-06-21:
I'll look into the logs. Here is my modified compose file https://github.com/chrisdeely/balena-pihole/blob/no-tailscale/docker-compose.yml

chrisdeely, 2023-06-21:
*sigh* this was all me - the admin page is at `/admin` and going to `/` gives a 403... not super helpful, but the system is working fine. Thanks for your quick responses, @klutchell !

closed 2023-06-21 by chrisdeely