klutchell/balena-pihole #252: Pi-hole v6 will be an android-based container

klutchell, 2023-09-24:
https://github.com/pi-hole/docker-pi-hole/issues/1430#issuecomment-1732666533

We should be prepared to make compatibility changes.

eiddor, 2023-09-25:
[image]

Alpine or Android?

eiddor, 2023-09-25:
If it's Alpine, it shouldn't be a big hairy deal - If it's Android, then...

klutchell, 2023-09-25:
I was on my phone, it must have auto-corrected.

Yeah, I don't think we will need to change much except some of the packages we install, and the commands we run to set console fonts.

eiddor, 2023-09-25:
Ok whew! Is it worth messing with the upstream Dockerfile and testing now or should we just wait until it actually happens?

klutchell, 2023-09-25:
I think just wait, we are in no rush and it will be much easier knowing with confidence what we are working with.

klutchell, 2023-11-30:
Good to know! If I get some time this holiday season I might draft an example PR with the required changes to support v6.

klutchell, 2023-12-01:
@DL6ER are we also moving away from s6-overlay by the looks of it? I can go look up the exact changes myself when I dig into it but I high level view of the planned changes would be handy.

DL6ER, 2023-12-01:
@promofaux is the `docker` master, he'd be the best to answer any questions

PromoFaux, 2023-12-01:
Master is one word for it. Fake it till you make it I guess!

Yeah, s6-overlay is gone. We took the opportunity to do a complete overhaul - basing on alpine has been a long time goal, so it was a good time to start from scratch!

klutchell, 2025-02-19:
Closed via https://github.com/klutchell/balena-pihole/pull/297

closed 2025-02-19 by klutchell