klutchell/balena-pihole #265: Configure Tailscale to use Balena Pihole as the DNS server for devices connected to Tailscale VPN

chrisfasnacht, 2024-01-01:
Today I setup Balena Pi Hole and noticed that it has Tailscale support. It says for accessing the pihole admin UI from outside the network.

I posted this in https://forums.balena.io/t/balenapihole-and-tailscale-vpn/369618 I'm not sure if there or here is the best place.

> Tailscale
> Included is a Tailscale service in order to [access your Pi-hole from anywhere](https://tailscale.com/kb/1114/pi-hole/).

I enabled PUBLIC DEVICE URL from the balena device summary screen and I can access the admin UI from outside of my network.

The Tailscale Pi hole link seems to suggest that I could use tailscale from a device to use pihole to block ads via the tailscale VPN.

> [Step 3: Set your Raspberry Pi as your DNS server ](https://tailscale.com/kb/1114/pi-hole#step-3-set-your-raspberry-pi-as-your-dns-server)
> 
> You can configure DNS for your entire Tailscale network from Tailscale’s [admin console](https://login.tailscale.com/admin/dns). Go to the [**DNS**](https://login.tailscale.com/admin/dns) page and enter your Raspberry Pi’s Tailscale IP address as a global nameserver.

I'd like to do this, but I'm **not sure how to access the tailscale account that's been setup in this Balena Project.**

Is this possible to do?

If not could this be a feature request to be added to the project?

klutchell, 2024-01-02:
Great question! It had never occurred to me before but devices that join the open fleet via the link on balenaHub do not have an active Tailscale connection as we are not providing an auth key of any kind by default.

So to answer your question, in order to make your Pi-hole available on your personal Tailnet you would need to deploy this app to your own fleet via the deploy instructions, then provide the `TS_AUTH_KEY` env var.

cdfasnacht, 2024-01-02:
So I can't add TS_AUTH_KEY as a device variable?

Also the public IP seems to NOT function as DNS servers on my mac (when I use the IP as a DNS server, my internet doesn't work). Is this intended?

I have a tailscale account from a previous project and the IPs I get from it begin with 100... is it possible to know the tailscale IP for my balena device using this project?

klutchell, 2024-01-02:
If you joined the existing public fleet (by flashing the image, and not by creating your own fleet) you cannot add ANY environment variables. I suggest you deploy to your own fleet using the [button on the README](https://github.com/klutchell/balena-pihole?#getting-started) then you can provide `TS_AUTH_KEY` as an environment variable on your fleet or device.

If you already deployed a fleet with the above project then you can go ahead and add the `TS_AUTH_KEY` env var with your own Tailnet API key. Then your Pi-hole will appear in your Tailscale Admin dashboard.

> Also the public IP seems to NOT function as DNS servers on my mac (when I use the IP as a DNS server, my internet doesn't work). Is this intended?

Which public IP are you referring to here? By default Pi-hole should be listening on all local interfaces of your Raspberry Pi device. I would strongly advise against opening those ports in your router.