klutchell/balena-pihole #324: PADD issues with v6

eiddor, 2025-02-21:
I haven't tried to troubleshoot yet, but my display is showing:

`"PADD only works in conjunction with Pi-hole!"`

eiddor, 2025-02-21:
We're failing here:

```bash
    # If setupVars.conf is not present, then PADD is not running on a Pi-hole
    # and we are not able to start as StartupRoutine() will fail below
    if [ ! -f /etc/pihole/setupVars.conf ]; then
      printf "%b" "${check_box_bad} Error!\n    PADD only works in conjunction with Pi-hole!\n"
      exit 1
```

It seems that `setupVars.conf` isn't being created:

```bash
pihole:/etc/pihole# ls -al setup*
-rw-rw---- 1 pihole pihole 400 Dec  2 17:14 setupVars.conf.update.bak
```

klutchell, 2025-02-21:
Weird that PADD would depend on that, because according to the comments that file is expected to be moved during migration.

https://github.com/pi-hole/docker-pi-hole/blob/29b604af3f1e65523dbb0971749efed236e151fa/src/bash_functions.sh#L178

Did the pihole process log anything helpful before PADD failed?

Note that we start pihole in the background then start polling for the API to become available before we start PADD, so if PADD thinks pihole isn't there then why did the FTL API respond?

eiddor, 2025-02-21:
Nothing out of the ordinary - And Pihole seems to be working just fine!

```
 pihole  Waiting for FTL API to be available...
 pihole  Setting console font to ter-u20n...
 pihole  Starting PADD...
 pihole    [i] Version info:
 pihole        Core version is v6.0.3 (Latest: v6.0.3)
 pihole        Web version is v6.0.1 (Latest: v6.0.1)
 pihole        FTL version is v6.0.1 (Latest: v6.0.1)
 pihole  
 pihole  2025-02-21 15:19:03.360 CST [65M] INFO: ########## FTL started on pihole! ##########
```

But here we go: https://github.com/pi-hole/PADD/issues/421

closed 2025-02-23 by klutchell