klutchell/balena-pihole #369: Padd service repeatedly restarting

hard-coded, 2025-08-06:
Device type: Raspberry Pi 4 (using 64bit OS)
OS version: 6.5.48+rev3
Flashed with Etcher via Balena Cloud
Deployed using one-click "Deploy with Balena" via https://github.com/klutchell/balena-pihole, release 2025.7.1

I had this error prior. I ended up starting from scratch, re-flashing, and redeploying and still get the same error.

```
2025-08-06T18:06:43+00:00 Restarting service 'padd sha256:4ac020227b36b837e640327b540bfeef66ff67220c1a329c6341f39cba935280'
2025-08-06T18:06:43+00:00  padd  Stopping plymouth service...
2025-08-06T18:06:43+00:00  padd  method return time=1754503603.730995 sender=:1.0 -> destination=:1.128 serial=1852 reply_serial=2
2025-08-06T18:06:43+00:00  padd     object path "/org/freedesktop/systemd1/job/3550"
2025-08-06T18:06:43+00:00  padd  Setting console font to ter-u16n...
2025-08-06T18:06:43+00:00  padd  setfont: ERROR kdfontop.c:212 put_font_kdfontop: Unable to load such font with such kernel version
2025-08-06T18:06:43+00:00  padd  Exited with code 71, sleeping for a few seconds...
2025-08-06T18:06:42+00:00 Service exited 'padd sha256:4ac020227b36b837e640327b540bfeef66ff67220c1a329c6341f39cba935280'
```

wesleymccloy, 2025-08-11:
I'm facing the same issue on balenaOS 6.5.48:
```
uname -a 
Linux e55b842 6.12.32-v8-16k #1 SMP PREEMPT Thu Jun  5 10:46:00 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux
```
I don't have a screen connected.

klutchell, 2025-08-12:
I've reproduced this on my Pi 4, but I don't have a screen for that device, so I can't be sure if this behaviour is related to the lack of attached screen or just kernel compatibility issues with the setfont command.

Does anyone have a Pi4 with a screen attached?

eiddor, 2025-08-12:
I do!

eiddor, 2025-08-13:
So, I missed the Pi4 detail and the only one of those that I have left is the one I can't upgrade for some reason.  It's sitting in the PiTFT case and would require a bit of surgery to swap out the microSD.

@klutchell Do feel up to helping me figure out why I can't change OS versions on it so I can do a proper test?  It's the same board that we've been beating on for years and it should be otherwise healthy.

klutchell, 2025-08-13:
@eiddor sure, enable support access and send me the UUID and I can take a look at it

eiddor, 2025-08-13:
Done and done.

closed 2025-08-14 by eiddor

wesleymccloy, 2025-08-15:
I’ve configured this particular service to not restart automatically since I’m not running a display at the moment.